zephyrproject-rtos/zephyr #30808: Blueooth: Unable start scanning

LingaoM, 2020-12-17:
**Describe the bug**

Use nrf52832 as controller, use nrf52840 as host, use H4 transport layer.

*** Booting Zephyr OS build zephyr-v2.4.0-1367-g7d16ab50d738  **

[00:00:02.807,769] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:02.807,800] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:02.807,800] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.4 Build 99
[00:00:02.825,592] <inf> bt_hci_core: Identity: e4:0c:5e:e9:04:9a (random)
[00:00:02.825,592] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:02.825,622] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
[00:00:09.433,471] <dbg> bt_mesh_health_cli.health_cli_init: primary 1
[00:00:09.433,776] <inf> bt_mesh_prov_device: Device UUID: 3a2b2d51-f1a3-47ad-bb49-7c23c85219b3
[00:00:09.438,903] <wrn> bt_hci_core: opcode 0x200b status 0x0c
[00:00:09.438,903] <err> bt_mesh_adv: starting scan failed (err -5)

I did not modify any code in the host or controller. This problem occurs from time to time. However, if the host and controller are in the same chip, the problem is not found.

cvinayak, 2020-12-22:
Please provide details to reproduce like the commit, samples, configuration, build steps used, flashing steps and test procedure.

carlescufi, 2021-01-07:
@LingaoM could you please pinpoint the exact HCI command that is getting command disallowed? This could be the set random address one, which would be different from what you analyzed.

carlescufi, 2021-01-14:
@LingaoM could you please provide additional information here?

LingaoM, 2021-01-15:
Sorry for the delay, I think it may be that the controller is not reset, and the host is running on another board, resulting in an inconsistent state.

carlescufi, 2021-01-21:
> Sorry for the delay, I think it may be that the controller is not reset, and the host is running on another board, resulting in an inconsistent state.

The Host should always start with an HCI Reset command after a Host reset, and that should soft-reset the controller. Any other additional information you can provide?

LingaoM, 2021-01-25:
Sorry, i can't provide more effective information.

LingaoM, 2021-01-25:
It is worth mentioning that I use auto pts to capture this problem, and it is probabilistic. If the host / controller is on the same chip, the same problem is not found.

carlescufi, 2021-02-04:
>[00:00:09.438,903] bt_mesh_adv: starting scan failed (err -5)

This has likely been fixed by #31532, please reopen if that's not the case.

closed 2021-02-04 by carlescufi