zephyrproject-rtos/zephyr #30810: tests: kernel: kernel.threads.armv8m_mpu_stack_guard fails on nrf9160dk

PerMac, 2020-12-17:
**Describe the bug**
The test from tests/kernel/threads/thread_stack/ fail on nrf9160dk
git bisect: c770cab1a3a6cbfefbe8432939f78b321770d498 is the first bad commit

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf 9160dk connected
2. go to zephyr dir
3. run `./scripts/sanitycheck --device-testing -T tests/kernel/threads/thread_stack/ -p nrf9160dk_nrf9160 --device-serial /dev/ttyACM0 -v -v`
4. See error

**Expected behavior**
Test passes 

**Impact**
Not clear

**Logs and console output**
```
DEBUG   - DEVICE: testing stest_member_stack
DEBUG   - DEVICE: Stack object 0x200000a0[544]
DEBUG   - DEVICE: - Testing supervisor mode
DEBUG   - DEVICE: - Thread reports buffer 0x200000c0 size 512
DEBUG   - DEVICE: - check read/write to stack buffer
DEBUG   - DEVICE: - Carved-out space in buffer: 0
DEBUG   - DEVICE: - Unused objects space: 0
DEBUG   - DEVICE: self-reported unused stack space: 124
DEBUG   - DEVICE: target thread unused stack space: 124
DEBUG   - DEVICE: PASS - test_stack_buffer
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_idle_stack
DEBUG   - DEVICE: hi! bye!
DEBUG   - DEVICE: E: could not allocate kernel object, out of memory
DEBUG   - DEVICE: E: ***** MPU FAULT *****
DEBUG   - DEVICE: E:   Data Access Violation
DEBUG   - DEVICE: E:   MMFAR Address: 0x8
DEBUG   - DEVICE: E: r0/a1:  0x00000000  r1/a2:  0x00000001  r2/a3:  0x00000001
DEBUG   - DEVICE: E: r3/a4:  0x00000000 r12/ip:  0xaaaaaaaa r14/lr:  0x0000074f
DEBUG   - DEVICE: E:  xpsr:  0x61000000
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x0000cf62
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x200002e0 (unknown)
DEBUG   - DEVICE: E: Halting system
DEBUG   - Waiting for device nrf9160dk_nrf9160 to become available
DEBUG   - Waiting for device nrf9160dk_nrf9160 to become available
```

**Environment (please complete the following information):**
 - Ubuntu 18.04
 - Toolchain  Zephyr SDK

ioannisg, 2021-01-21:
Looking into it now

ioannisg, 2021-01-22:
@nashif this is probably fixed by the PR I sent earlier, #31539. I aimed at fixing the exact same MPU_FAULT I am seeing here (I came across the same test failure on qemu mps2_an521)

ioannisg, 2021-01-23:
Fixed by #31539

closed 2021-01-23 by ioannisg