zephyrproject-rtos/zephyr #30819: twister: --generate-hardware-map crashes and deletes map

erwango, 2020-12-17:
**Describe the bug**
`twister --generate-hardware-map ../map.yaml` returns error and deletes map when a new board is detected.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug a new board to the bench
2. Run `twister --generate-hardware-map ../map.yaml`
3. See reported error:
```
Renaming output directory to /local/mcu/zephyrproject/zephyr/twister-out.12
INFO    - Scanning connected hardware...
WARNING - Unsupported device (None): /dev/ttyS4 - n/a
Traceback (most recent call last):
  File "/local/mcu/zephyrproject/zephyr/scripts/twister", line 1269, in <module>
    main()
  File "/local/mcu/zephyrproject/zephyr/scripts/twister", line 797, in main
    hwm.save(options.generate_hardware_map)
  File "/local/mcu/zephyrproject/zephyr/scripts/pylib/twister/twisterlib.py", line 3985, in save
    yaml.dump(hwm, yaml_file, Dumper=Dumper, default_flow_style=False)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/__init__.py", line 290, in dump
    return dump_all([data], stream, Dumper=Dumper, **kwds)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/__init__.py", line 278, in dump_all
    dumper.represent(data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 27, in represent
    node = self.represent_data(data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 48, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 199, in represent_list
    return self.represent_sequence('tag:yaml.org,2002:seq', data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 92, in represent_sequence
    node_item = self.represent_data(item)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 52, in represent_data
    node = self.yaml_multi_representers[data_type](self, data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 343, in represent_object
    'tag:yaml.org,2002:python/object:'+function_name, state)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 118, in represent_mapping
    node_value = self.represent_data(item_value)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 52, in represent_data
    node = self.yaml_multi_representers[data_type](self, data)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/yaml/representer.py", line 317, in represent_object
    reduce = data.__reduce_ex__(2)
  File "/usr/lib/python3.6/multiprocessing/sharedctypes.py", line 198, in __reduce__
    assert_spawning(self)
  File "/usr/lib/python3.6/multiprocessing/context.py", line 356, in assert_spawning
    ' through inheritance' % type(obj).__name__
RuntimeError: Synchronized objects should only be shared between processes through inheritance
```
4. map.yaml is still there but empty.

**Expected behavior**
New board is populated in map.yaml

**Impact**
showstopper: test bench could not run anymore

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04
 - Toolchain Zephyr SDK (0.12.0 Beta)
 - zephyr-v2.4.0-2539-gb2b8554da8

**Additional context**
Add any other context about the problem here.

erwango, 2020-12-17:
@nashif Hit the issue once again on my set up and it seems we're facing the same issue on our bench.
From what I can see this is finally not an error from my part.

nashif, 2020-12-17:
yeah, I have  a fix for this already. will push asap

PR #30823 (2020-12-20): twister: fix hardware map load/save
Do not store shared memory variables into yaml and check for validity of
imported hardware map before attempting to modify it.

Fixes #30819
closed 2020-12-20 by nashif