zephyrproject-rtos/zephyr #30841: How to disable reception of socket CAN frames

mejtejt, 2020-12-17:
**Is your feature request related to a problem? Please describe.**
I am using setsockopt function to subscribe to the messages I want tor receive on the CAN. The function is as follows:
```
setsockopt(socket_id, SOL_CAN_RAW, CAN_RAW_FILTER, &filter,
                       sizeof(filter));
```
And the code runs, I am receiving messages I want. After some time, I want to tell socket CAN "stop receiving those messages". I tried implementing it with the following function:

`setsockopt(socket_id, SOL_CAN_RAW, CAN_RAW_FILTER, NULL,0);`

But the function call fails, with a reason: Invalid argument. So, my question is: how can I tell socket CAN to stop receiving messages on a socket?

jukkar, 2020-12-17:
Indeed the code checks that option and size are correct.
See https://github.com/zephyrproject-rtos/zephyr/blob/ae35d3000f73e5ff104d4277e64103b11da2b092/subsys/net/lib/sockets/sockets_can.c#L603 and few lines below for details.
If giving NULL option and/or 0 length would mean that the filter needs to be unregistered, then these checks should be changed. Would you be able to send patches for these?

jukkar, 2020-12-17:
BTW, have you checked how the filter is removed in Linux? We should probably do it similarly in Zephyr.

mejtejt, 2020-12-18:
Setting optval to NULL, and optlen to 0 iz exactly how I use it on Linux. The following is an excerpt from my Linux code:

[image]

And it works just like I want it to. This is part of the CAN_Open function, where I want to explicitly set that no messages are to be  received before instructed to do so.

closed 2021-03-22 by github-actions[bot]