zephyrproject-rtos/zephyr #30850: iotdk: couldn't flash image into iotdk board using west flash.

chen-png, 2020-12-18:
**Describe the bug**
Couldn't use west flash command to flash image into iotdk board.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b iotdk -p auto samples/hello_world/
2. west flash
3. see error

**Logs and console output**

-- west flash: using runner openocd
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-06-25-12:07)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter
  adapter assert |deassert [srst|trst [assert|deassert srst|trst]]
  adapter deassert |assert [srst|trst [deassert|assert srst|trst]]
  adapter usb
    adapter usb location [<bus>-port[.port]...]
adapter_khz [khz]
adapter_name
adapter_nsrst_assert_width [milliseconds]
adapter_nsrst_delay [milliseconds]
interface driver_name
interface_list
reset_config [none|trst_only|srst_only|trst_and_srst]
          [srst_pulls_trst|trst_pulls_srst|combined|separate]
          [srst_gates_jtag|srst_nogate] [trst_push_pull|trst_open_drain]
          [srst_push_pull|srst_open_drain]
          [connect_deassert_srst|connect_assert_srst]
**Error: invalid subcommand "driver ftdi"
zephyrproject/zephyr/boards/arc/iotdk/support/openocd.cfg:11: Error:
in procedure 'script'
at file "embedded:startup.tcl", line 26
at file "zephyrproject/zephyr/boards/arc/iotdk/support/openocd.cfg", line 11**
FATAL ERROR: command exited with status 1: zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s zephyrproject/zephyr/boards/arc/iotdk/support -s  zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f zephyrproject/zephyr/boards/arc/iotdk/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'load_image zephyrproject/zephyr/build/zephyr/zephyr.elf' -c 'resume 0x0000000020001354' -c shutdown

**Environment (please complete the following information):**
 - OS: fedora32
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: ae35d3000f

nashif, 2020-12-18:
@abrodkin  @ruuddw fyi

abrodkin, 2020-12-18:
Update SDK to v0.12.0 which was finally released yesterday: https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.12.0.

chen-png, 2020-12-21:
it was passed with sdk 0.12.0, closed this.

closed 2020-12-21 by chen-png