zephyrproject-rtos/zephyr #30860: legacy timeout ticks mishandled

pabigot, 2020-12-18:
This provides an issue backing #30842 which by inspection is correct and appears to be high priority.  With legacy timeouts `Z_TIMEOUT_TICKS` converts the tick count to a timeout which is expressed in milliseconds, not ticks.

Ticks should be assigned directly to timeout value in case of
CONFIG_LEGACY_TIMEOUT_API=y, just as they were before referenced patch.

Fixes: 7a815d5d993d ("kernel: sched: Use k_ticks_t in z_tick_sleep")

Tested with:
```
	LOG_INF("uptime: %d", (int) k_uptime_get());
	k_sleep(K_SECONDS(1));
	LOG_INF("uptime: %d", (int) k_uptime_get());
```
in `ztest` framework.

Before this PR:
```
I: uptime: 2070400
I: uptime: 2080500
```
after:
```
I: uptime: 207040
I: uptime: 208050
```

andyross, 2020-12-18:
@pabigot posted this along with the fix, so it's for him.

But... the legacy timeout feature was a source compatibility layer for a feature that landed and became default three (four?) releases ago.  Maybe the correct fix is just to remove the feature?  If no one has tripped over this in legacy code in the past year, no one will.  And no one should be writing new code to it by default.

closed 2020-12-18 by nashif

andyross, 2020-12-18:
I misassigned it anyway, was @mniestroj 's patch.  But it's closed not.   I should have read through the cluster of issues and comments first.  Note that as a root cause, this is really a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/30893

PR #30842 (2020-12-18): kernel: sched: fix legacy timeout calculation in z_tick_sleep
Ticks should be assigned directly to timeout value in case of
CONFIG_LEGACY_TIMEOUT_API=y, just as they were before referenced patch.

Due to: 7a815d5d993d ("kernel: sched: Use k_ticks_t in z_tick_sleep")
Fixes: #30860

Tested with:
```
	LOG_INF("uptime: %d", (int) k_uptime_get());
	k_sleep(K_SECONDS(1));
	LOG_INF("uptime: %d", (int) k_uptime_get());
```
in `ztest` framework.

Before this PR:
```
I: uptime: 2070400
I: uptime: 2080500
```
after:
```
I: uptime: 207040
I: uptime: 208050
```