zephyrproject-rtos/zephyr #30861: drivers: uart: increase timeout precision in uart_rx_enable

nordic-krch, 2020-12-18:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, `uart_rx_enable` gets `timeout` given in milliseconds. `timeout` refers to inactivity period which leads to `UART_RX_RDY` event. Millisecond resolution may significantly lower communication throughput if protocol requires ACK. For example, device transmits 10 bytes @1Mbaud, it takes ~100us but receiver has 1ms timeout so packet will be received after ~1100us. Another case is when hardware has inactivity detection.

**Describe the solution you'd like**
Change timeout to microseconds. Optionally, add inline function or macro for converting characters to microseconds.

** Alternative **
Change timeout to characters. It is baudrate independent and may  be suitable for hardware inactivity detection.

shlomow, 2020-12-18:
Maybe it is worth to add  more definitions like `SYS_FOREVER_MS` to the api that have soft restrictions on the drivers implementation as well as the occurrence of the `UART_RX_RDY` signal. For example, we can add `WAIT_INACTIVITY` define that that may be implemented by drivers as 1 ms timeout, and can be implemented using some hardware advantages. In the define I described the `UART_RX_RDY` will be received when the driver find inactivity (very not formal definition) and the reception won't stop as the buffer is not full yet.

nordic-krch, 2021-10-11:
`k_timeout_t` was rejected because it is to kernel bound. Some drivers may use hardware timeout or require higher precision than system clock and use other source for calculating timeout, e.g. counter device.

closed 2021-10-12 by carlescufi