zephyrproject-rtos/zephyr #30862: Nordic system timer driver incompatible with LEGACY_TIMEOUT_API

pabigot, 2020-12-18:
At some point the implementation started depending on support for absolute timeouts: attempts to build hello world with `CONFIG_LEGACY_TIMEOUT_API=y` produce errors.

The "fix" for this may be to stop supporting `LEGACY_TIMEOUT_API`.  #30860 shows that a breaking change to that implementation can go almost two weeks without being noticed.

mniestroj, 2020-12-18:
I've noticed that issue yesterday after bumping Zephyr in one of old downstream projects. In most cases that doesn't happen, as projects tend to rely on stable releases instead. So that breaking change could be noticed even after 2.5.0 would be released.

Anyway, the problem comes from incomplete testing of `CONFIG_LEGACY_TIMEOUT_API=y` I guess.

pabigot, 2020-12-18:
@nashif Should we plan to remove LEGACY_TIMEOUT_API for LTS?

nashif, 2020-12-18:
> @nashif Should we plan to remove LEGACY_TIMEOUT_API for LTS? AFAICT it was introduced at the start of 2.3 by #23382

yes, we should.

nashif, 2020-12-18:
> @nashif Should we plan to remove LEGACY_TIMEOUT_API for LTS? AFAICT it was introduced at the start of 2.3 by #23382

> yes, we should.

we need to create another issue for that though..

nashif, 2020-12-18:
@pabigot  what is the reason you are still using the legacy timeouts with nordic timer?

pabigot, 2020-12-18:
I'm not; I was trying to validate #30842 and had to use NXP because Nordic doesn't build hello world with that option.    Also see https://github.com/zephyrproject-rtos/zephyr/issues/30862#issuecomment-748049838 for why this might show up in the real world.

andyross, 2020-12-18:
Obviated by https://github.com/zephyrproject-rtos/zephyr/pull/30903 FWIW

nashif, 2021-01-05:
duplicate of #30893

closed 2021-01-05 by nashif