zephyrproject-rtos/zephyr #30918: up_squared:  tests/kernel/mem_protect/mem_protect failed.

chen-png, 2020-12-21:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/mem_protect/mem_protect
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
 Booting Zephyr OS build zephyr-v2.4.0-2666-gfe8311843285  
Running test suite memory_protection_test_suite

START - test_permission_inheritance
E: Page fault at address 0x12e990 (error code 0x4)
E: Linear address not present in page tables
E: Access violation: user thread not allowed to read
E: PTE: not present
E: RAX: 0x0000000000000000 RBX: 0x0000000000000000 RCX: 0x0000000000100456 RDX: 0x0000000000000000
E: RSI: 0x0000000000d60fb4 RDI: 0x0000000000d79378 RBP: 0x0000000000000000 RSP: 0x0000000000d60fc8
E:  R8: 0x000000000010048c  R9: 0x8000000000000006 R10: 0x0000000000000000 R11: 0x0000000000000202
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000d60fc8 RFLAGS: 0x0000000000010202 CS: 0x003b CR3: 0x000000000083a000
E: RIP: 0x0000000000100499
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x12a940 (unknown)
Caught system error -- reason 0 1
fatal error expected as part of test case
 PASS - test_permission_inheritance

START - test_inherit_resource_pool
 PASS - test_inherit_resource_pool

**START - test_mem_domain_setup**
E: architecture-specific initialization failed for domain 0x12bf20 with -12
E: RAX: 0x0000000000000004 RBX: 0x0000000000000002 RCX: 0x0000000000000001 RDX: 0x0000000000000046
E: RSI: 0x000000000000000a RDI: 0x000000000012e890 RBP: 0x0000000000d74fa0 RSP: 0x0000000000d74f60
E:  R8: 0x0000000000000001  R9: 0x0000000000d74d2d R10: 0x00000000ffffffff R11: 0x0000000000000031
E: R12: 0x000000000012bf20 R13: 0x0000000000000002 R14: 0x0000000000000212 R15: 0x0000000000000000
E: RSP: 0x0000000000d74f60 RFLAGS: 0x0000000000000046 CS: 0x0018 CR3: 0x000000000117d000
E: RIP: 0x0000000000110063
**E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x12d720 (ztest_thread)
Caught system error -- reason 4 0
fatal error was unexpected, aborting**

**Environment (please complete the following information):**
 - OS: Fedora32
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: fe8311843285

s-kelvin, 2020-12-22:
After bisect, I found that previously the timeout issue in test_permission_inheritance has been fixed by 8ad5ad22140611cc335380bd078dc627ec4412b9, so before that commit, the results of testcases following test_permission_inheritance are unknown due to timeout. 

@jenmwms mentioned this unexpected fatal error in test_mem_domain_setup and found the bad commit in [#30305](https://github.com/zephyrproject-rtos/zephyr/issues/30305#issuecomment-737593923). Could you please advise and help?

enjiamai, 2020-12-24:
@jocelyn-li  @s-kelvin 
I did some analysis on this, the test case failed due to k_mem_domain_init() failed. 

test_mem_domain_setup() ->  k_mem_domain_init() -> arch_mem_domain_init().
In arch/x86/core/x86_mmu.c:1623  

domain->arch.ptables = page_pool_get();   it returns -ENOMEM.
This means no more page pool can be used.

And if we look at boards/x86/up_squared/Kconfig.defconfig:31  
config X86_MMU_PAGE_POOL_PAGES
        default 3092 if X86_MMU

If we set X86_MMU_PAGE_POOL_PAGES > 3584 + 24,  the test case pass.

dcpleung, 2021-01-27:
Could you re-test?

nashif, 2021-01-27:
cant reproduce

closed 2021-01-27 by nashif

chen-png, 2021-01-28:
tested with the latest code, it has been fixed.