zephyrproject-rtos/zephyr #30921: west flash failed with an open ocd error

manoj153, 2020-12-21:
**Describe the bug**
I was trying to test my first blinky sample and encountered a error 

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr
2. west build -p auto -b nucleo_h723zg samples/basic/blinky
3. See error

**Impact**
Not allowed to flash and test a starter project to get started
 
**Logs and console output**
`
<pre><font color="#4E9A06"><b>manoj@M-Ping</b></font>:<font color="#3465A4"><b>~/zephyrproject/zephyr</b></font>$ west flash
<font color="#8AE234">-- west flash: rebuilding</font>
[0/1] cd /home/manoj/zephyrproject/zephyr/build/zephyr/cmake/flash &amp;&amp; /usr/bin/cmake -E echo

<font color="#8AE234">-- west flash: using runner openocd</font>
<font color="#8AE234">-- runners.openocd: Flashing file: /home/manoj/zephyrproject/zephyr/build/zephyr/zephyr.hex</font>
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-06-25-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
init
Info : STLINK V3J7M2 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.286400
Info : clock speed 1800 kHz
Info : stlink_dap_op_connect(connect)
Info : SWD DPIDR 0x6ba02477
Info : STM32H723ZG.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : STM32H723ZG.cpu0: external reset detected
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080027c0 msp: 0x24050000
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0  STM32H723ZG.ap2    mem_ap     little STM32H723ZG.cpu    halted
 1* STM32H723ZG.cpu0   cortex_m   little STM32H723ZG.cpu    halted

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080027c0 msp: 0x24050000
Warn : Cannot identify target as a STM32H7xx family.
Error: auto_probe failed
auto erase enabled
<font color="#EF2929">FATAL ERROR: command exited with status 1: /home/manoj/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/manoj/zephyrproject/zephyr/boards/arm/nucleo_h723zg/support -s /home/manoj/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/manoj/zephyrproject/zephyr/boards/arm/nucleo_h723zg/support/openocd.cfg -c init -c targets -c &apos;reset halt&apos; -c &apos;flash write_image erase /home/manoj/zephyrproject/zephyr/build/zephyr/zephyr.hex&apos; -c &apos;reset halt&apos; -c &apos;verify_image /home/manoj/zephyrproject/zephyr/build/zephyr/zephyr.hex&apos; -c &apos;reset run&apos; -c shutdown</font>
</pre>

`
**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 
 - zephyr-v2.4.0-2571-g553a6e34a8

**Additional context**
Add any other context about the problem here.

manoj153, 2020-12-23:
Issue solved, found out because of the outdated open-OCD package, for now, I upgraded to latest v0.12.0 package

closed 2020-12-23 by manoj153

luludamermaid, 2021-07-31:
how did you install openocd v0.12.0 SDK package? what's the command line?

manoj153, 2021-09-09:
change the binary?