zephyrproject-rtos/zephyr #30929: PDM Driver for nrf52840dk

nehalrohit, 2020-12-21:
I want to interface with an MP34DT05 MEMS audio sensor for nrf52840dk_pca10056.

can you give me some steps or any sample example? 
Thanks...
Nehal

anangl, 2021-01-04:
@nehalrohit You can take a look at #27671 and perhaps use it as your starting point. This PR aims at adding API for PDM drivers so any feedback in this regard is welcome.

nehalrohit, 2021-01-04:
Hi @anangl 
I  added all files and their config for nrf52840dk 
do the same as the change did for Thingy:52 and create pdm example.
I have below problem 
[image]

closed 2021-03-20 by github-actions[bot]