zephyrproject-rtos/zephyr #30935: tests: net: sockets: tcp: add a tls tests

cfriedt, 2020-12-21:
**Is your enhancement proposal related to a problem? Please describe.**
Currently I'm attempting to use mbedTLS in a custom application and am creating a test to verify TLS in CI. The test is multi-threaded and uses TCP sockets with the TLS extensions; the server server socket is managed by one thread and the client socket is managed by the main thread.

My reference(s) for the application were the `echo_server` / `echo_client` sample applications and I'm just using the certificates / keys from those sample applications as-is. However, I'm experiencing a handshake error in TLS.

**Describe the solution you'd like**
It might be useful to have it a test that verified this kind of functionality in Zephyr CI

**Describe alternatives you've considered**
Have looked around for existing tests that exercise this particular setup, but have found none.

**Additional context**
Current handshake error:
```
*** Booting Zephyr OS build zephyr-v2.4.0-2547-gb68f47fdb9a0  ***
W: No entropy device on the system, TLS communication may be insecure!
I: Initializing network
I: Initializing built-in certificates
D: Adding CA Certificate (783 bytes)
D: Adding Server Certificate (Public Key) (693 bytes)
D: Adding Server Certificate (Private Key) (1219 bytes)
...
D: (main): Allocated TLS context, 0x2000189c
D: (main): socket: ctx=0x20005840, fd=1
I: CPort 0 mapped to TCP/IP port 4242
D: (main): Allocated TLS context, 0x20001c74
D: (main): socket: ctx=0x200058c8, fd=3
I: CPort 1 mapped to TCP/IP port 4243
...
D: (main): Allocated TLS context, 0x2000204c
D: (main): socket: ctx=0x20005950, fd=5
D: (rx_q[0]): parent=0x200058c8, ctx=0x200059d8, st=0
D: (main): there are 1 tags
D: (main): tag[0]: 0
D: (main): found tag
E: returning tls_add_ca_certificate()
E: calling mbedtls_x509_crt_parse(0x200021b4, 0x2e047, 783)
D: (main): found a CA cert
D: (rx_q[0]): ctx=0x200059d8, pkt=0x2000ba60, st=0, user_data=(nil)
D: (greybus): accept: ctx=0x200059d8, fd=7
D: (greybus): Allocated TLS context, 0x20002424
D: (greybus): there are 1 tags
D: (greybus): tag[0]: 1
D: (greybus): found tag
E: returning tls_set_own_cert()
E: tls_set_credential() failed (-22)
E: tls_mbedtls_set_credentials() failed (-22)
E: tls_mbedtls_init() failed (-22)
D: (greybus): close: ctx=0x200059d8, fd=7
D: (greybus): discarding pkt 0x2000ba60
D: (rx_q[0]): ctx=0x20005950, pkt=(nil), st=-104, user_data=(nil)
D: (rx_q[0]): Marked socket 0x20005950 as peer-closed
D: (main): NULL return from fifo
E: TLS handshake error: -7280
ASSERTION FAIL [r == 0] @ CMAKE_SOURCE_DIR/src/gpio.c:147
	connect: 113
E: r0/a1:  0x00000004  r1/a2:  0x00000093  r2/a3:  0x90000000
E: r3/a4:  0x0002880b r12/ip:  0x00000000 r14/lr:  0x00000c6f
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x0001ca2a
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20001300 (main)
E: Halting system
qemu-system-arm: terminating on signal 2
ninja: build stopped: interrupted by user.
```

jukkar, 2020-12-22:
How this is different from #30936, do we need two TLS issues that look identical?

cfriedt, 2021-01-06:
Looks line I made a duplicate issue.

closed 2021-01-06 by cfriedt