zephyrproject-rtos/zephyr #30938: samples/net/dhcpv4_client does not work with sam_e70_xplained

pabigot, 2020-12-21:
Building this sample for `sam_e70_explained` reveals two problems:
* By default the MAC address is 00:00:00:00:00:00.  This can be eliminated by adding `CONFIG_ETH_SAM_GMAC_MAC_I2C_EEPROM=y` somewhere.
* Once that's fixed wireshark shows that the device is not transmitting any DHCP requests.

The same sample with `frdm_k64f` works as expected.

jukkar, 2020-12-22:
Works for me (after enabling the MAC address)

```
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_2 initialized
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_0 initialized
uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-2666-gfe8311843285  ***
[00:00:00.001,000] <inf> eth_sam: MAC: fc:c2:3d:11:c1:47
[00:00:00.001,000] <inf> eth_sam: Queue 0 activated
[00:00:00.001,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.001,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
[00:00:00.091,000] <inf> eth_sam_phy: PHYID: 0x221561 at addr: 0
[00:00:00.097,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:03.122,000] <inf> eth_sam: Link up
[00:00:06.027,000] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:07.030,000] <inf> net_dhcpv4: Received: 192.168.178.84
[00:00:07.030,000] <inf> net_dhcpv4_client_sample: Your address: 192.168.178.84
[00:00:07.030,000] <inf> net_dhcpv4_client_sample: Lease time: 864000 seconds
[00:00:07.030,000] <inf> net_dhcpv4_client_sample: Subnet: 255.255.255.0
[00:00:07.030,000] <inf> net_dhcpv4_client_sample: Router: 192.168.178.1
```

pabigot, 2020-12-22:
Huh.  Wonder if it's a board problem.  Mine has on the back A09-2637/02 sn 12000009780. On the front there's a JX02 94V09 26 18 where the last two suggest it may have been built in July 2018.

jukkar, 2020-12-22:
> Mine has on the back A09-2637/02

I have earlier version A09-2637/01 from 2017. I remember that there was two versions (A vs B) of this board, perhaps this is related to board version. I though the difference between the boards was related to number of TSN queues, and not about IP connectivity.

pabigot, 2020-12-22:
Aha.  Yes, it turns out I do have the B variant, as indicated by this cryptic output:

    [00:00:00.000,000] <wrn> soc: CIDR mismatch: chip = 0xa1020e01 vs HAL = 0xa1020e00

When I switch to `sam_e70b_xplained` it works.

closed 2020-12-22 by pabigot