zephyrproject-rtos/zephyr #30964: Sleep calls are off on qemu_x86

GroteGnoom, 2020-12-22:
**Describe the bug**
When running the dining philosophers sample on qemu_x86, the philosophers don't sleep when eating, and they should.

So, for example, this philosopher does not sleep for 450 ms:
```
Philosopher 0 [P: 0]   EATING  [  450 ms ] 
```

The generated build.ninja has a `-icount shift=5,align=off,sleep=off` in the qemo-system-i386 command, and when `sleep=on` the problem is fixed. The origin of this seems to be `/zephyr/cmake/emu/qemu.cmake`.

**To Reproduce**
```
west build -b qemu_x86 samples/philosophers
west build -t run
```

**Expected behavior**
I expected
```
k_msleep(delay);
```
to work. Any kind of sleep calls seem to sleep far shorter than the indicated time.

**Impact**
Annoyance. The [readme philosophers example](https://docs.zephyrproject.org/latest/samples/philosophers/README.html) does not work as intended.

**Environment:**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.11.4
 - Commit SHA or Version used: commit 201a8d29cdb2f856f7f889f9a75d3a463e3491ef

**Further info:**
Maybe this was introduced here: https://github.com/zephyrproject-rtos/zephyr/issues/14173. It is unclear to me why `sleep` should be `off`.

GroteGnoom, 2020-12-22:
Ah, I only searched the open issues, my bad. This has been reported here as well:
https://github.com/zephyrproject-rtos/zephyr/issues/29518
Shouldn't this be fixed in general, instead of everybody removing their icounts?

nashif, 2021-01-05:
closing as a non-bug.

closed 2021-01-05 by nashif