zephyrproject-rtos/zephyr #30977: stm32f4 error in adc register size in dtsi files. Missing adc nodes from stm32f405.dtsi

zisisadamos, 2020-12-23:
**Describe the bug**
stm32f4.dtsi incorrectly states that adc1 node starts at 0x40012000 with size 0x400. Although that [RM0368](https://www.st.com/resource/en/reference_manual/dm00096844-stm32f401xb-c-and-stm32f401xd-e-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf#page=39&zoom=100,165,493) states that adc spans from 0x40012000 to 0x400123FF if we take a closer look at adc register mapping [here](https://www.st.com/resource/en/reference_manual/dm00096844-stm32f401xb-c-and-stm32f401xd-e-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf#page=240&zoom=100,165,493) we can see that we have slots for multiple adcs that are reserved and their size is 0x4d.

**To Reproduce**
Compare stm32f4.dtsi against [this](https://www.st.com/resource/en/reference_manual/dm00096844-stm32f401xb-c-and-stm32f401xd-e-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf#page=240&zoom=100,165,493)

Compare stm32f405.dtsi against [this](https://www.st.com/resource/en/reference_manual/dm00031020-stm32f405-415-stm32f407-417-stm32f427-437-and-stm32f429-439-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf)

**Expected behavior**
Expected to have adc1 node in stm32f4.dtsi having reg property equal to <0x40012000 0x4d>
Expected to have adc2 and adc3 in stm32f405.dtsi having reg properties equal to  <0x40012100 0x4d>  <0x40012200 0x4d> accordingly.

**Impact**
Showstopper

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK
 - 92ecc7866cbe02815f66e4783f044ca7ee22284a

FRASTM, 2021-01-04:
Correct, 
Note that in RefMan RM0090 and RM0383 for stm32f411 the ADC1 is mapped from 0x000 - **0x04C** in the ADC area (0x4001 2000 - 0x4001 23FF)

zisisadamos, 2021-01-04:
So I have added the missing nodes. But I cannot add the interrupts because python script complains. Is there any work-around for that?

FRASTM, 2021-01-04:
Which python script fails ?
I can still compile the  `tests/drivers/adc/adc_api`/  for the  nucleo_f401re board after changing the adc area of the stm32f4.dtsi

zisisadamos, 2021-01-04:
**gen_isr_tables.py: error: multiple registrations at table_index 18 for irq 18 (0x12)
Existing handler 0x80259e5, new handler 0x80259e5** . Thats the error if i modify stm32f405.dtsi by adding 

`adc2: adc@40012100 {
			compatible = "st,stm32-adc";
			reg = <0x40012100 0x4d>;
			clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00000100>;
			interrupts = <18 0>;
			status = "disabled";
			label = "ADC_2";
			#io-channel-cells = <1>;
		};` 

and 

`		adc3: adc@40012200 {
			compatible = "st,stm32-adc";
			reg = <0x40012200 0x4d>;
			clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00000100>;
			interrupts = <18 0>;
			status = "disabled";
			label = "ADC_3";
			#io-channel-cells = <1>;
		};`
If i declare them as nodes on my board dts then I get the error I mentioned in the beggining.

FRASTM, 2021-01-04:
Ok I see.
BTW : STM32_CLOCK_BUS_APB2 0x00000200  for adc2 and STM32_CLOCK_BUS_APB2 0x00000400 for adc3

zisisadamos, 2021-01-04:
Is that correctly classified as a bug? Since I am missing the nodes? Is it something that I dont see?

FRASTM, 2021-01-04:
Well, not really a bug. This is the zephyr interrupt handler configuration which limits to only **one** active (in the board dts) at a time.
All ADC instances can share the same irq vector (18) but cannot be all activated simultaneously.

If you need more than one ADC, maybe a global ADC dtsi is better (with large reg = <0x40012000 0x400>) instead of splitting in 3 ADC areas as you only have one interrupt handler (and you may miss the 'Common' register bank).

windelbouwman, 2021-01-14:
This would be good to add!

My suggestion:
- Add 3 register ranges of size 0x4D, one for each ADC (compatible with sm32-adc)
- Add a ADC common register range

Open issue: how to deal with the common register settings?

GunZi200, 2021-04-19:
I suppose this issue is related to: https://github.com/zephyrproject-rtos/zephyr/issues/26874

erwango, 2021-11-03:
Closing in favor of #26874

closed 2021-11-03 by erwango