zephyrproject-rtos/zephyr #30997: samples: net: sockets: echo_client: posix example

cfriedt, 2020-12-25:
**Is your enhancement proposal related to a problem? Please describe.**
It would be good to provide a sample application to run on POSIX systems, linking with OpenSSL, to connect to the `echo_sever` sample application (running on hardware or in qemu) with TLS enabled.

**Describe the solution you'd like**
Provide a source file in `samples/net/sockets/echo_client/src` as well as a `Makefile.posix` that can be used to create an `echo_client` binary for Linux.

**Describe alternatives you've considered**
It might be as simple as using the usual [TLS Client](https://wiki.openssl.org/index.php/SSL/TLS_Client) example code, but perhaps it might be a bit more nuanced. Some things to consider would be exercising `TLS_PEER_VERIFY_REQUIRED` as well just to demonstrate how mutual authentication should be done.

**Additional context**

jukkar, 2020-12-25:
The net-tools project contains `echo-client` program that can be used from Linux. The https://github.com/zephyrproject-rtos/net-tools/blob/master/README.md file contains information how to use stunnel to tunnel TLS traffic between Linux and zephyr.

cfriedt, 2020-12-26:
Interesting - that might be useful for testing. My eventual use case will likely not be able to use stunnel though.

cfriedt, 2020-12-27:
I suspect that the existing solution of using `stunnel` will be sufficient for Zephyr & most people's testing needs, but just in case anyone else needs  a decent reference for a suitable client (and server) for Linux that uses OpenSSL, they can use the attached. It's MIT licensed, tested, etc.

I wrote a test client / server because the application-specific use case required it.

The attached code uses the certificates and keys in `/samples/sockets/net/echo_server/src`, so `ca.der`, `server.der`, and `server_privkey.der`.

For tests with invalid certificates (signed with some other CA cert?), it uses certificates and keys in `samples/net/sockets/dumb_http_server_mt/src`, so `mt-http-server-cert.der` and `mt-http-server-key.der`.

There are tests for one-way authentication and mutual authentication and tests to ensure that missing or invalid certs throw (sorry, it's C++). Feel free to include these examples in `net-tools`, but I doubt it's 100% necessary.

```
[==========] Running 6 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 6 tests from TLSTest
[ RUN      ] TLSTest.tls_peer_verify_none
[       OK ] TLSTest.tls_peer_verify_none (116 ms)
[ RUN      ] TLSTest.tls_peer_verify_optional_no_client_cert
[       OK ] TLSTest.tls_peer_verify_optional_no_client_cert (113 ms)
[ RUN      ] TLSTest.tls_peer_verify_optional_client_cert
[       OK ] TLSTest.tls_peer_verify_optional_client_cert (120 ms)
[ RUN      ] TLSTest.tls_peer_verify_required_no_client_cert
SSL_get_peer_certificate() failed
[       OK ] TLSTest.tls_peer_verify_required_no_client_cert (14 ms)
[ RUN      ] TLSTest.tls_peer_verify_required_invalid_client_cert
140575324714816:error:14094418:SSL routines:ssl3_read_bytes:tlsv1 alert unknown ca:../ssl/record/rec_layer_s3.c:1543:SSL alert number 48
SSL_accept() failed (-1)
140575324710656:error:1417C086:SSL routines:tls_process_client_certificate:certificate verify failed:../ssl/statem/statem_srvr.c:3701:
[       OK ] TLSTest.tls_peer_verify_required_invalid_client_cert (14 ms)
[ RUN      ] TLSTest.tls_peer_verify_required
[       OK ] TLSTest.tls_peer_verify_required (114 ms)
[----------] 6 tests from TLSTest (492 ms total)

[----------] Global test environment tear-down
[==========] 6 tests from 1 test suite ran. (492 ms total)
[  PASSED  ] 6 tests.
```

[openssl-tls-example-20201227-1609086170.tar.xz.txt](https://github.com/zephyrproject-rtos/zephyr/files/5744780/openssl-tls-example-20201227-1609086170.tar.xz.txt)

closed 2020-12-27 by cfriedt