zephyrproject-rtos/zephyr #31008: error: initializer element is not constant .attr = K_MEM_PARTITION_P_RX_U_RX

alexisc67, 2020-12-26:
**Describe the bug**
Find an error when building a basic sample Blinky for the Thingy:52.
Here is the error:
zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/mpu/arm_core_mpu.c:77:3: error: initializer element is not constant
   .attr = K_MEM_PARTITION_P_RX_U_RX,
   ^
zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/mpu/arm_core_mpu.c:77:3: error: (near initialization for 'static_regions[0].attr')

**To Reproduce**
Steps to reproduce the behavior:
1. Run in the command line: west build -p auto -b thingy52_nrf52832 samples/basic/blinky

**Expected behavior**
Complete build

**Impact**
showstopper

**Logs and console output**
zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/mpu/arm_core_mpu.c:77:3: error: initializer element is not constant
   .attr = K_MEM_PARTITION_P_RX_U_RX,
   ^
zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/mpu/arm_core_mpu.c:77:3: error: (near initialization for 'static_regions[0].attr')

**Environment (please complete the following information):**
 - MacOS
 - Toolchain GNUARMEMB
 - Zephyr version: 2.4.99

**Additional context**
Add any other context about the problem here.
[image]

pabigot, 2020-12-26:
Which version of ARM embedded are you using?  I'm unable to reproduce this on Linux at current master head 4ca7942411 with either gcc-arm-none-eabi-9-2020-q2-update or gcc-arm-none-eabi-10-2020-q4-major.

ioannisg, 2021-01-04:
will take a look

ioannisg, 2021-01-25:
@alexisc67 I cannot reproduce this either, at current master (7b9b3cfe5a129580ec97c40e7a98d2fab07fa78c)

carlescufi, 2021-01-26:
Unable to reproduce, closing. Feel free to reopen if this is still an issue.

closed 2021-01-26 by carlescufi