zephyrproject-rtos/zephyr #31018: up_squared: tests/kernel/pipe/pipe_api failed.

chen-png, 2020-12-28:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/pipe/pipe_api
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
Booting Zephyr OS build zephyr-v2.4.0-2674-g4ca7942411f7
Running test suite pipe_api

**START - test_pipe_thread2thread**
E: General protection fault (code 0x0)
E: RAX: 0x0000000000cac000 RBX: 0x0000000000000000 RCX: 0x0000000000000000 RDX: 0x0000000000000000
E: RSI: 0xffffffffffffffff RDI: 0x0000000000118420 RBP: 0x0000000000d42fa8 RSP: 0x0000000000d42f18
E:  R8: 0x0000000000100be1  R9: 0x8000000000000006 R10: 0x0000000000000000 R11: 0x0000000000000246
E: R12: 0x0000000000118420 R13: 0xffffffffffffffff R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000d42f18 RFLAGS: 0x0000000000010286 CS: 0x003b CR3: 0x0000000000cac000
E: RIP: 0x0000000000100784
**E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x118060 (unknown)
E: Halting system**

**Environment (please complete the following information):**
 - OS: Fedora32
 - Toolchain: zephyr-sdk-0.12.0
 - Commit ID: 4ca7942411f7

nashif, 2021-01-05:
cant reproduce:

```
*** Booting Zephyr OS build zephyr-v2.4.0-2768-ge33f756f0e30  ***
Running test suite pipe_api
===================================================================
START - test_pipe_thread2thread
 PASS - test_pipe_thread2thread
===================================================================
START - test_pipe_put_fail
 PASS - test_pipe_put_fail
===================================================================
START - test_pipe_get_fail
 PASS - test_pipe_get_fail
===================================================================
START - test_half_pipe_get_put
 PASS - test_half_pipe_get_put
===================================================================
START - test_pipe_reader_wait
 PASS - test_pipe_reader_wait
===================================================================
START - test_pipe_avail_r_lt_w
 PASS - test_pipe_avail_r_lt_w
===================================================================
START - test_pipe_avail_w_lt_r
 PASS - test_pipe_avail_w_lt_r
===================================================================
START - test_pipe_avail_r_eq_w_full
 PASS - test_pipe_avail_r_eq_w_full
===================================================================
START - test_pipe_avail_r_eq_w_empty
 PASS - test_pipe_avail_r_eq_w_empty
===================================================================
START - test_pipe_avail_no_buffer
 PASS - test_pipe_avail_no_buffer
===================================================================
Test suite pipe_api succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

chen-png, 2021-01-06:
it failed with sdk 0.12.0, worked with sdk 0.11.4.

andyross, 2021-02-03:
Fix up.  This one was fun.  The proximate cause that fingered the new toolchain was that gcc is now doing some pretty fancy autovectorization of code and using SSE instructions in places it wasn't earlier.  And these were crashing on misaligned loads because our stack alignment for userspace threads was incorrect.

PR #31955 (2021-02-03): arch/x86: Fix stack alignment for user threads
The x86_64 SysV ABI requires 16 byte alignment for the stack pointer
during execution of normal code.  That means that on entry to an
ABI-compatible C function (which is reached via a CALL instruction
that pushes the return address) the RSP register must be MISaligned by
exactly 8 bytes.  The kernel mode thread setup got this right, but we
missed the equivalent condition in userspace entry.

The end result was a misaligned stack, which is surprisingly robust
for most use.  But recent toolchains have starting doing some more
elaborate vectorization [... cut]
commit cce5ff15 (2021-02-03): arch/x86: Fix stack alignment for user threads

The x86_64 SysV ABI requires 16 byte alignment for the stack pointer
during execution of normal code.  That means that on entry to an
ABI-compatible C function (which is reached via a CALL instruction
that pushes the return address) the RSP register must be MISaligned by
exactly 8 bytes.  The kernel mode thread setup got this right, but we
missed the equivalent condition in userspace entry.

The end result was a misaligned stack, which is surprisingly robust
for most use.  But recent toolchains have starting doing some more
elaborate vectorizatio [... cut]
closed 2021-02-03 by nashif