zephyrproject-rtos/zephyr #31019: Bluetooth: Mesh: Thread competition leads to failure to open or close the scanning.

LingaoM, 2020-12-28:
**Describe the bug**
Bluetooth Mesh unable to start scanning or disable scanning, as a result, the provisioning procedure failed.

Use NRF52DK as Controller attach to Linux OS, Run Zephyr Bluetooth Host on Qemu.

**To Reproduce**
Steps to reproduce the behavior:
`/* Controller */`
1. west build -b nrf52dk_nrf52832 zephyr/samples/bluetooth/hci_uart/
2. west flash
3. sudo btattach -B /dev/ttyACM1 -S 115200 -R and Expecting Manufacturer: 1521.
4. sudo btproxy -u -z -i 1

`/* Host */`
1. west build -b qemu_cortex_m3 zephyr/tests/bluetooth/tester/
2. west build -t menuconfig -> enable console to uart print, enable logging, enable mesh debug.
``` C
	struct bt_le_scan_param scan_param = {
			.type       = BT_HCI_LE_SCAN_PASSIVE,
			.filter_dup = BT_LE_SCAN_OPT_NONE,
			.interval   = MESH_SCAN_INTERVAL,
			.window     = MESH_SCAN_WINDOW };
	int err;

	BT_INFO("");
```
3. I adjusted the scanned log in the adv.c file in the mesh directory from DEBUG to INFO
4. west build

`/* auto-pts */`
1. ./autoptsclient-zephyr.py "PATH\NAME.pqw6" ~/zephyrproject/build/zephyr/zephyr.elf -i 192.168.120.1 -l 192.168.109.133 -c MESH/NODE/PROV/BV-12-C

**Logs and console output**
Attachment has been submitted, see iut-zephyr.zip
[iut-zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/5746104/iut-zephyr.zip)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used : zephyr-v2.4.0-1367-g7d16ab50d738

**Additional context**
Add any other context about the problem here.

LingaoM, 2020-12-28:
Based on the log information, it is preliminarily inferred that it is a competition issue caused by the ADV thread and the HCI RX thread simultaneously enabling or closing scanning.

https://github.com/zephyrproject-rtos/zephyr/blob/380528676960d0fb57fae48d8616154a5e06705c/subsys/bluetooth/host/hci_core.c#L8563-L8574

ADV Thread will also operate scanning when closing the broadcast. At this time, this kind of competition problem may occur when the HCI RX thread calls `bt_mesh_start`-->`bt_mesh_scan_start`

``` C
[00100050] <dbg> bt_hci_core.bt_buf_get_cmd_complete: sent_cmd 0x2000ab28
[00100050] <dbg> bt_hci_core.hci_cmd_complete: opcode 0x200c
[00100050] <dbg> bt_hci_core.hci_cmd_done: opcode 0x200c status 0x00 buf 0x2000ab28
[00100050] <dbg> bt_hci_core.bt_hci_cmd_send_sync: rsp 0x2000ab28 opcode 0x200c len 1
[00100050] <dbg> bt_hci_core.set_random_address: 37:6c:ea:5b:ea:45
[00100050] <dbg> bt_hci_core.bt_hci_cmd_create: opcode 0x2005 param_len 6
[00100050] <dbg> bt_hci_core.bt_hci_cmd_create: buf 0x2000ab28
[00100050] <dbg> bt_hci_core.bt_hci_cmd_send_sync: buf 0x2000ab28 opcode 0x2005 len 9
[00100050] <dbg> bt_hci_core.send_cmd: Sending command 0x200c (buf 0x2000ab40) to driver
[00100050] <dbg> bt_hci_core.bt_send: buf 0x2000ab40 len 5 type 0
[00100050] <inf> bt_driver: 0c20020000
[00100050] <dbg> bt_hci_core.process_events: ev->state 0
[00100050] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 2 events
[00100050] <dbg> bt_hci_core.process_events: count 2
[00100050] <dbg> bt_hci_core.process_events: ev->state 4
[00100050] <dbg> bt_hci_core.send_cmd: calling net_buf_get
[00100050] <dbg> bt_hci_core.send_cmd: calling sem_take_wait
[00100050] <dbg> bt_hci_core.bt_buf_get_cmd_complete: sent_cmd 0x2000ab40
[00100050] <dbg> bt_hci_core.hci_cmd_complete: opcode 0x200c
[00100050] <dbg> bt_hci_core.hci_cmd_done: opcode 0x200c status 0x0c buf 0x2000ab40
[00100050] <dbg> bt_hci_core.send_cmd: Sending command 0x2005 (buf 0x2000ab28) to driver
[00100050] <dbg> bt_hci_core.bt_send: buf 0x2000ab28 len 9 type 0
[00100050] <inf> bt_driver: 05200645ea
[00100050] <dbg> bt_hci_core.process_events: ev->state 0
[00100050] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 2 events
[00100050] <wrn> bt_hci_core: opcode 0x200c status 0x0c
[00100050] <err> bt_mesh_adv: stopping scan failed (err -5)
```
This test item(MESH/NODE/PROV/BV-12-C) will repeatedly provision and reset nodes, the problem is in the reset process so that capture the problem.

LingaoM, 2020-12-28:
https://github.com/zephyrproject-rtos/zephyr/blob/380528676960d0fb57fae48d8616154a5e06705c/subsys/bluetooth/mesh/cfg_srv.c#L1967-L1998

I think if it is possible to put the` bt_mesh_reset` operation in `bt_mesh_send_cb::end` to execute, there will be no such problem.

But the problem now is that the sending of the proxy does not implement the real callback. I think it is meaningful to implement the callback at this time.This will remove the concept of `bt_mesh_proxy_on_idle` because it is too obscure, which will fix the current problem and also optimize the existing network (for example, the publish operation will be based on periodicity when proxy used). @jhedberg @trond-snekvik  Any thoughts to this ?

Finally: https://github.com/zephyrproject-rtos/zephyr/blob/380528676960d0fb57fae48d8616154a5e06705c/subsys/bluetooth/mesh/pb_gatt.c#L121-L131
This is also a potential problem, because the callback is not implemented. As a result, certain specific provision procedure method cannot be performed normally.

carlescufi, 2021-01-07:
From the log:
>[00081960] <wrn> bt_hci_core: opcode 0x200c status 0x0c

So Set Scan Enable is failing with Command Disallowed, which is different from what is seen in #30808, where it was a Set Scan Parameters command that failed.

@cvinayak how can Set Scan Enable fail with command disallowed at all?

cvinayak, 2021-01-07:
@carlescufi The controller does not return success if scanning is already disabled. This is kept so, because the Zephyr Host maintains flags and it should not go out of sync w.r.t the state of scanning. This can be fixed in controller to ignore disables on already disabled scanning, but if host is maintaining incorrect state, then enabling an already started scanning by incorrect host state would fail with command disallowed.

As the host sets the `BT_DEV_SCANNING` flag [1] before the command complete in `set_le_scan_enable_legacy`  and `set_le_ext_scan_enable`, if there is a `bt_le_scan_stop` call made by another thread (while waiting for command complete), which in turn calls `bt_le_scan_update`, would there be incorrect flags state for `BT_DEV_SCANNING` and also based on the race condition as to which thread gets the hci_cmd buffer from `bt_hci_cmd_create`, controller may be asked to disable [2] a non-started scanning state?

[1] https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L1086
[2] https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L4202

carlescufi, 2021-01-14:
@trond-snekvik will reproduce and try to find out what sequence in the code is causing a second Set Scan Enable to be sent.

trond-snekvik, 2021-01-21:
I think Lingao's proposed fix in #31038 will resolve the issue for this specific case in the mesh, but there's still an issue in the general bluetooth.h interface.

The mesh is not using the bluetooth.h API wrong, and this can still occur in other bluetooth applications even after the mesh change is merged:

As Lingao proved in the comment above, the problem happens because the mesh is disabling advertising from one thread, then disabling scanning from another, before the advertising disable has exited.

The sequence of events is:
1. The controller receives an advertisement, and enqueues it for host processing.
2. In the host, the Mesh Adv Thread calls `bt_le_adv_stop()`
   - hci_core calls `set_le_adv_enable_legacy(DISABLE)`, which waits for the controller to respond by pending on its `sync_sem`.
   - The controller handles the command, and enqueues the response for host processing
3. hci_core starts processing its event queue
   - It finds the advertisement from step 1, and passes it up to the mesh stack.
      - The mesh stack realizes that this is a reset() message, and calls `bt_mesh_reset()`, which disables the scanner
      - The mesh stack calls `bt_le_scan_stop()`, which calls a `bt_le_scan_update()`, which calls `set_le_scan_enable(DISABLE)`.
      - hci_core sends the 0x200c scan disable HCI command, then pends on its `sync_sem` for the response
      - The controller handles the command, and enqueues the response for host processing
   - hci_core processes the adv_disable response from step 2.
      - It resumes processing of `bt_le_adv_stop()`, which checks the `BT_DEV_SCANNING` flag, which is still set.
      - It calls `set_le_scan_enable(DISABLE)` to regenerate the private addr.
      - **This fails in the controller, since it is already disabled.** The controller enqueues the failure response.
   - hci_core processes the first scan_disable response and clears the `BT_DEV_SCANNING` flag.
   - hci_core processes the second scan_disable response, and reports the error.

TL;DR: in the end of `bt_le_adv_stop()`, we're not checking if a scan command is already pending, and we end up doing a double disable.

This is the same sort of issue we've seen in hci_core multiple times before, where the flag state is only changed after a successful command, but someone tries to read the flag and send the same command while the controller is processing the first command.

carlescufi, 2021-01-21:
For clarification, according to the spec both enabling the scanning when already enabled and disabling the scanner when already disabled shall return success:

```
If the LE_Scan_Enable parameter is set to 0x01 and scanning is already
enabled, any change to the Filter_Duplicates setting shall take effect.
Disabling scanning when it is disabled has no effect.
```

carlescufi, 2021-01-21:
Plan forward:
1. @cvinayak fix the controller so it complies with the spec as per my comment above
2. @trond-snekvik and @joerchan to analyze if other instances of the sequence described [here](https://github.com/zephyrproject-rtos/zephyr/issues/31019#issuecomment-764576774) are potentially present in the Host.

cvinayak, 2021-01-22:
@LingaoM Please verify if #31532 works around the failure to enable/disable scanning.

carlescufi, 2021-01-28:
> Plan forward:
>
> 1. @cvinayak fix the controller so it complies with the spec as per my comment above
> 2. @trond-snekvik and @joerchan to analyze if other instances of the sequence described [...]

PR is posted, @trond-snekvik will verify that #31532, and then 2. is still remaining.

PR #31532 (2021-01-29): Bluetooth: controller: Fix accept scan en/disable if already en/disabled
Conform to Bluetooth Specification, if the LE_Scan_Enable
parameter is set to 0x01 and scanning is already enabled,
any change to the Filter_Duplicates setting shall take
effect. Disabling scanning when it is disabled has no
effect.

Fixes #31019.
commit 1afbe077 (2021-01-29): Bluetooth: controller: Fix accept scan en/disable if already en/disabled

Conform to Bluetooth Specification, if the LE_Scan_Enable
parameter is set to 0x01 and scanning is already enabled,
any change to the Filter_Duplicates setting shall take
effect. Disabling scanning when it is disabled has no
effect.

Fixes #31019.
closed 2021-01-29 by nashif