zephyrproject-rtos/zephyr #31020: CI build failed when submitted a PR on intel_adsp_cavs18

KangJianX, 2020-12-28:
**Describe the bug**
Submit a mailbox test cases on Github, CI build failed on adsp_cavs18

**To Reproduce**
Push a PR about mailbox test cases.

**See error**
......
[ 64%] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_common.c.obj
[ 65%] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/nanosleep.c.obj
[ 67%] Linking C static library liblib__posix.a
[ 67%] Built target lib__posix
Scanning dependencies of target intel_adsp_common
[ 67%] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/adsp.c.obj
[ 68%] Building ASM object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/main_entry.S.obj
[ 69%] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/soc.c.obj
[ 69%] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/soc_mp.c.obj
[ 70%] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/trace_out.c.obj
[ 71%] Linking C static library libintel_adsp_common.a
[ 71%] Built target intel_adsp_common
Scanning dependencies of target subsys__testsuite__ztest
[ 72%] Building C object zephyr/subsys/testsuite/ztest/CMakeFiles/subsys__testsuite__ztest.dir/src/ztest.c.obj
[ 72%] Linking C static library libsubsys__testsuite__ztest.a
[ 72%] Built target subsys__testsuite__ztest
Scanning dependencies of target drivers__ipm
[ 73%] Building C object zephyr/drivers/ipm/CMakeFiles/drivers__ipm.dir/ipm_cavs_idc.c.obj
[ 74%] Building C object zephyr/drivers/ipm/CMakeFiles/drivers__ipm.dir/ipm_intel_adsp.c.obj
/home/ztest/work/zephyrproject/zephyr/drivers/ipm/ipm_intel_adsp.c:13:10: fatal error: platform/mailbox.h: No such file or directory
   13 | #include <platform/mailbox.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
zephyr/drivers/ipm/CMakeFiles/drivers__ipm.dir/build.make:75: recipe for target 'zephyr/drivers/ipm/CMakeFiles/drivers__ipm.dir/ipm_intel_adsp.c.obj' failed
make[2]: *** [zephyr/drivers/ipm/CMakeFiles/drivers__ipm.dir/ipm_intel_adsp.c.obj] Error 1
CMakeFiles/Makefile2:2965: recipe for target 'zephyr/drivers/ipm/CMakeFiles/drivers__ipm.dir/all' failed
make[1]: *** [zephyr/drivers/ipm/CMakeFiles/drivers__ipm.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04.4 LTS
 - SDK: zephyr-sdk-0.12.0
 - Commit 380528676960d0fb57fae48d8616154a5e06705c

s-kelvin, 2020-12-29:
Same problem here.

yingmingx, 2020-12-29:
Same problem here,too.

andyross, 2021-01-05:
Confirmed.  That's a SOF header being pulled in instead of the (AFAICT) identical one in the Zephyr tree, and that fails when you aren't building the SOF app.

Really a better scheme here would be to clean up that driver.  Right now we have a Zephyr IPM driver (which is our own oddball IPC mechanism) built on top of the platform headers, and the core arch layer in the kernel then uses it to implement the low-level IPI mechanism for SMP.  But nothing else uses nor cares about IPM.

What we should have is a in-zephyr-tree declaration of the relevant registers, implement IPI on top of that, and then have a side hook for a stub IPM driver for applications that actually care.

KangJianX, 2021-01-07:
> Confirmed. That's a SOF header being pulled in instead of the (AFAICT) identical one in the Zephyr tree, and that fails when you aren't building the SOF app.
>
> Really a better scheme here would be to [...]

So, do you have an ETA for this issue?

nashif, 2021-01-12:
this should be fixed already.

closed 2021-01-12 by nashif