zephyrproject-rtos/zephyr #31027: Google tests run twice

mejtejt, 2020-12-28:
**Describe the bug**
I am unit testing my C++ code with google test framework in Zephyr. When, inside of the tests folder prj.conf file, I enable 
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y

tests are executed twice. 

**Expected behavior**
I want tests to run once. 

**Logs and console output**
The following screen shot represents the structure of my project:
[image]

The following is the content of CMakeLists.txt file in test folder. 

```
cmake_minimum_required(VERSION 3.13.1)

set(CMAKE_CXX_COMPILER /usr/bin/g++ CACHE PATH "" FORCE)

set(CMAKE_C_FLAGS -m32)
set(CMAKE_CXX_FLAGS -m32)

set(BOARD native_posix)
set(ZEPHYR_TOOLCHAIN_VARIANT zephyr)
set(TOOLCHAIN_ROOT ${ZEPHYR_BASE})
set(ZEPHYR_EXTRA_MODULES $ENV{ZEPHYR_BASE}/../greyp-zephyr-packages/modules/googletest
                         $ENV{ZEPHYR_BASE}/../greyp-zephyr-packages/modules/nanopb
                        )

include_directories("../impl/src/greyp_connectivity"
                    "../impl/src/greyp_messages"
                    "../impl/src/conversion_utilities"
                    "../impl/src/serial_driver"
                    "../impl/src/micro_messaging"
                    $ENV{ZEPHYR_BASE}/../greyp-zephyr-packages/build
                    )

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(cyklaer_test)

FILE(GLOB app_sources ../impl/src/greyp_connectivity/*.cpp
                      ../impl/src/greyp_messages/*.cpp
                      ../impl/src/micro_messaging/*.cpp
                      )
FILE(GLOB test_sources src/*.cpp)
target_sources(app PRIVATE ${test_sources} ${app_sources})
target_link_libraries(app PUBLIC gtest_main)
```

At this point, prj.conf file in test folder is empty. Running the tests results in the following output:
[image]

However, when in test prj.conf file, I add 

**Additional context**
Add any other context about the problem here.

mejtejt, 2021-01-07:
There is another piece of information I came across, which I also do not understand why is it happening, and for which I think it is a pre-requisite for C++ configs to work. 

In in test/prj.conf file, instead of having 

```
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
```

I put 

```
CONFIG_EXTERNAL_LIBC=n
CONFIG_NEWLIB_LIBC=y
```

and try to build that, I get the following output:

[image]

And I cannot figure out why is always `external lib` taken as C library. How do I instruct it to use `new lib` instead?

pabigot, 2021-01-12:
Are you using `native_posix` as the target board?  That's probably the only environment where gtest is going to function at all.  For that board you can't select newlib.

pabigot, 2021-01-12:
One likely cause:  Zephyr initializes C++ static objects during the kernel startup code, rather than when the C runtime infrastructure iterates over them before starting.  If gtest uses static objects (probably does) then they may be being initialized twice, which might cause them to run twice.

I don't think gtest is likely to work well with C++ in Zephyr, where C++ is not a first-class citizen.

mejtejt, 2021-01-19:
Yes, I am using `native_posix` as target board. So basically, the only way to keep using gtest and C++ would be to somehow pass gtest_repeat = 1 from command line, before starting tests?

pabigot, 2021-01-19:
If that works, yes.  I've never used `gtest` with Zephyr.

pabigot, 2021-01-22:
I'm going to close this, because there's a workaround even though the use of gtest with Zephyr and C++ isn't really supported.  If there are objections it can be re-opened as a low-priority enhancement.

closed 2021-01-22 by pabigot

sidmodi-mw, 2023-11-09:
@mejtejt I am trying to integrate `gtest` into my Zephyr project and was hoping you could share some insight on how you integrated the library - did you use `gmock` as well? And were you able to get past this double running issue? I'm finding it really hard to find any resources on how to do this integration. Any pointers would be much appreciated!