zephyrproject-rtos/zephyr #31028: Cannot READ_BIT(RCC->CR, RCC_CR_PLL1RDY) on STM32H743 based board

othmaneAZ, 2020-12-28:
Hi,
I am working on a STM32H743 based custom board by porting it to Zephyr. I am trying to build a firmware for the blinky sample and this build runs just fine. After flashing the firmware on the board using the STLINK-v2, the sample does not run as expected. I managed to run a debug session using OpenOCD and I figured out that my program stops in the stm32h7xx_ll_rcc.h
`4352      return ((READ_BIT(RCC->CR, RCC_CR_PLL1RDY) == (RCC_CR_PLL1RDY))?1UL:0UL);`
I can't understand the root causes of this bug. 
I defined in my BOARD_defconfig file the clock configuration this way so I can get 400MHz as a frequency of the CPU and 200MHZ and 100MHz for the peripherals through AHBx and APBx buses. I can't see what I'm doing wrong.
```
# clock configuration
CONFIG_CLOCK_CONTROL=y

# Clock configuration for Cube Clock control driver
CONFIG_CLOCK_STM32_HSE_CLOCK=25000000
CONFIG_CLOCK_STM32_SYSCLK_SRC_PLL=y

# use HSE (bypass) as PLL input
CONFIG_CLOCK_STM32_PLL_SRC_HSE=y
CONFIG_CLOCK_STM32_HSE_BYPASS=y

# produce 400MHz clock at PLL output
CONFIG_CLOCK_STM32_PLL_M_DIVISOR=2
CONFIG_CLOCK_STM32_PLL_N_MULTIPLIER=64
CONFIG_CLOCK_STM32_PLL_P_DIVISOR=2
CONFIG_CLOCK_STM32_PLL_Q_DIVISOR=4
CONFIG_CLOCK_STM32_PLL_R_DIVISOR=2

```
Thank you,

gmarull, 2020-12-29:
From my experience HSE is usually generated by an actual crystal on custom boards, so `CONFIG_CLOCK_STM32_HSE_BYPASS` should be set to `n`. Not sure if that's the problem, but it may help.

othmaneAZ, 2020-12-30:
Thank you @gmarull, That actually helped me solve the bug. 
Do you know any documents that explain why HSE is usually generated by an actual crsytal and not an external user clock on custom boards ? Just to understand better. 
Thank you again !

gmarull, 2020-12-30:
Many ST dev boards use ST-Link clock output, this is why bypass is needed, on custom boards you'll likely have a crystal. Would you mind closing the issue?

othmaneAZ, 2020-12-30:
Thank you @gmarull, I will !

closed 2020-12-30 by othmaneAZ