zephyrproject-rtos/zephyr #31036: BMI 160 i2c version not working. I modified to i2c in kconfig to make it use i2c.

VinodSHegde, 2020-12-29:
Build failed.
bmi160.c : warning : 'bmi160_api' defined but not used [-Wunused-cons-variable=]

Apart from modifying SPI to I2C in Kconfig, where do I need to modify to make the program work for BMI160 on board NRF52840 using I2C

VinodSHegde, 2020-12-29:
#define BMI160_DEVICE_INIT(inst)					\
	DEVICE_DT_INST_DEFINE(inst, bmi160_init, device_pm_control_nop,	\
		&bmi160_data_##inst, &bmi160_cfg_##inst,		\
		POST_KERNEL, CONFIG_SENSOR_INIT_PRIORITY,		\
		&bmi160_api);

There is some errot in DEVICE_DT_INST_DEFINE

I changed into other definition where error goes but doesn't boot. What to do to get BMI160 i2c working with NRF52840DK?

VinodSHegde, 2021-01-01:
BMI 160 nrf_twim_error and failed to set PMU error.

I am unable to resolve any!

nashif, 2021-01-12:
@sjg20  can you take a look?

sjg20, 2021-01-13:
@VinodSHegde what are you trying to do? The driver should build for I2C automatically if you have an I2C bus on your system.

For native_posix I an using this to test it:

west build --board native_posix tests/drivers/sensor/generic/

I have not tried this driver on real i2c hardware as I don't have any that has that chip.

So I think there could well be bugs.

VinodSHegde, 2021-02-14:
1. After writing suspension for magnetometer :
do {
			if (bmi160_byte_read(dev, BMI160_REG_PMU_STATUS,
					     &sts.raw) < 0) {
				return -EIO;
			}

			if (i == 0) {
				pmu_set = (pmu_sts->mag == sts.mag);
			} else if (i == 1) {
				pmu_set = (pmu_sts->acc == sts.acc);
			} else {
				pmu_set = (pmu_sts->gyr == sts.gyr);
			}

		} while (!pmu_set);
pmu_set is always zero.
If I comment out do-while : 
then 
2. While setting {BMI160_CMD_PMU_ACC | pmu_sts->acc, 3200}, if (bmi160_byte_write(dev, BMI160_REG_CMD, cmds[i].cmd) < 0) {
			return -EIO;
		}
- function returns nothing and goes to infinite waiting until manually remove SCL or SDA pin.

sjg20, 2021-02-23:
Hi,

Is there a test for what you are writing?

If you look at the emulator you can see the features it supported. II don't think it supports setting values. You could perhaps add that feature?

Regards,
Simon

sjg20, 2021-04-01:
Closing for inactivity

closed 2021-04-01 by sjg20

rfc00, 2022-02-07:
> 1. After writing suspension for magnetometer :
>    do {
>    if (bmi160_byte_read(dev, BMI160_REG_PMU_STATUS,
>    &sts.raw) < 0) {
>    return -EIO;
>    }
>    ```
>     	if (i == 0) {
>     		pmu_set = [...]

Any solution found for this?