zephyrproject-rtos/zephyr #31043: Infinite loop in cmd_handler_process

jimparis, 2020-12-29:
In `drivers/modem/modem_cmd_handler.c`, there is potential for an infinite loop here:
https://github.com/zephyrproject-rtos/zephyr/blob/5358a11687a3a04a85f8c3b1f862bbb3b696221b/drivers/modem/modem_cmd_handler.c#L426-L429

Basically this will loop forever if it receives too much data without a newline.

My particular setup that triggers the bug:
I'm using a nRF52 with `drivers/wifi/esp.c` which uses `modem_cmd_handler`, with a baudrate change via the `target_speed` devicetree property.
What happens:

* ESP boots up at 115200 baud
* esp.c changes baudrate to 1000000
* Hardware glitch causes ESP to reboot, spitting out boot messages at 115200 baud.
* nRF receives this boot message at the wrong baudrate, so it's interpreted as a huge string of `\0`.
* Then it gets stuck in that loop:
    * `cmd_handler_process_iface_data` fills rx buffer
    * `cmd_handler_process_rx_buf` finds no CRLF in `findcrlf`, so it returns with the buffer unchanged
    * eventually the buffer fills up:
    ```[00:01:18.931,152] <err> modem_iface_uart: Rx buffer doesn't have enough space```
    * now `cmd_handler_process_iface_data` fails to allocate more space in the buffer, and returns immediately without even checking the UART: https://github.com/zephyrproject-rtos/zephyr/blob/5358a11687a3a04a85f8c3b1f862bbb3b696221b/drivers/modem/modem_cmd_handler.c#L282-L290
    * loop forever...

A hack that prevents the lockup is to set `alloc_timeout` so that a full buffer at least lets other threads run:
```diff --git a/drivers/wifi/esp/esp.c b/drivers/wifi/esp/esp.c
index ca89ecbe12e3..f799138cb5fc 100644
--- a/drivers/wifi/esp/esp.c
+++ b/drivers/wifi/esp/esp.c
@@ -1001,7 +1001,7 @@ static int esp_init(const struct device *dev)
        data->cmd_handler_data.match_buf = &data->cmd_match_buf[0];
        data->cmd_handler_data.match_buf_len = sizeof(data->cmd_match_buf);
        data->cmd_handler_data.buf_pool = &mdm_recv_pool;
-       data->cmd_handler_data.alloc_timeout = K_NO_WAIT;
+       data->cmd_handler_data.alloc_timeout = K_SECONDS(1);
        data->cmd_handler_data.eol = "\r\n";
        ret = modem_cmd_handler_init(&data->mctx.cmd_handler,
                                       &data->cmd_handler_data);
```
Thoughts for a proper fix: probably `cmd_handler_process` should have a return value, and return an error when the buffer fills up without finding CRLF.  Then the caller (like `esp.c:esp_rx`) can detect the error and reinitialize as desired.
Or, `cmd_handler_process` could just discard the buffer and try to continue; that wouldn't require changes in callers.
Maybe @mike-scott or @mniestroj have thoughts?

mniestroj, 2021-01-05:
@jimparis I think we should check whether buffer is full when `findcrlf()` inside `cmd_handler_process_rx_buf()` doesn't find anything. So `Let's exit and leave any data for next time` should be suplemented with `Let's check if there is more space in buffer and log. If not, log an error and drop everything we've got in the buffer.`. This will both throw a warning that something is wrong and prevent infinite loop you are observing.

```
		frag = NULL;
		/* locate next CR/LF */
		len = findcrlf(data, &frag, &offset);
		if (!frag) {
			/*
			 * No CR/LF found.  Let's exit and leave any data
			 * for next time
			 */
			break;
		}
```

carlescufi, 2021-03-23:
@jimparis is this something you will be able to submit a fix for?

jimparis, 2021-03-24:
Not at the moment, sorry.  Maybe in a few weeks when I return to this project and can test changes.

closed 2021-06-07 by github-actions[bot]