zephyrproject-rtos/zephyr #31049: 18830 and 18831 ipso objects (mqtt) support for lwm2m client

mmoyer2, 2020-12-30:
**Is your enhancement proposal related to a problem? Please describe.**
The zephyr samples have a mqtt and lwm2m (with ipso object) client examples but it would be useful to establish an mqtt connection via lwm2m ipso objects 18830 MQTT Broker and 18831 MQTT Publication.

**Describe the solution you'd like**
Add 18830 and 18831 ipso objects functionality and examples so that a mqtt broker and publication can be established using the lwm2m client sample.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

rlubos, 2021-01-07:
Just a note, this would require to implement `Corelnk` data type as well (which is part of LwM2M 1.1) as it's used by MQTT Publication object.

carlescufi, 2023-11-22:
There are hundreds of LWM2M object types. Unfortunately we cannot implement support for all of them without help. However, we will accept Pull Requests implementing those, so please feel free to contribute support.

closed 2023-11-22 by carlescufi