zephyrproject-rtos/zephyr #31079: Receiving extended scans on an Adafruit nRF 52840

StevenGhe, 2021-01-03:
**Describe the bug**
A clear and concise description of what the bug is.
The "bt_le_scan_start" method fails with error -5, when using extended advertisements (CONFIG_BT_EXT_ADV).
After investigating I've found the error " bt_hci_core: opcode 0x2041 status 0x01". This comes from hci_core.c, line 341, It returns an IO error. Not too sure what could be wrong with IO?

What have you tried to diagnose or workaround this issue?
I've ran the sample sync_adv with the same error. 
CONFIG_BT_EXT_ADV disabling solves the issue but won't scan extended advertisements obviously.
Tried with different Adafruit feather, same result.

**To Reproduce**
Steps to reproduce the behavior:
Since the error happens on the sample as well, run the sample bluetooth > periodic_sync.
Issue isn't related with any periodic advertising. Only starting the extended scanner.

**Expected behavior**
A clear and concise description of what you expected to happen.
Same behaviour as starting it without extended advertisements: Callbacks being invoked on each scan.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Blocker. Have a deadline to work with new Bluetooth features (ext advertisements)

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: Win 10
 - Toolchain Zephyr SDK
 -  newest version 2.4, 1e61d3eb9b

cvinayak, 2021-01-04:
did you turn on the experimental CONFIG_BT_CTLR_ADV_EXT=y ?

See, https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/bluetooth/periodic_sync/boards/nrf52840dk_nrf52840.conf

StevenGhe, 2021-01-04:
I did not. This fixed it for me! 
I didn't knew there weren't any board specific config's...

You just saved me a from a lot of frustration. :)

Tyvm

closed 2021-01-04 by StevenGhe