zephyrproject-rtos/zephyr #31087: Support for ram retention in nrf52840

AnanthJaisankar, 2021-01-04:
I want to keep some variables n my application retained even after system power off mode.Do we have support for ram retention for nrf52840?.

nashif, 2021-01-05:
@carlescufi  fyi

utzig, 2021-01-05:
Related to https://github.com/mcu-tools/mcuboot/issues/905

carlescufi, 2021-01-29:
Assignig to @anangl since he has a plan to provide a sample using memory retention.

pabigot, 2021-01-29:
@carlescufi I thought I was going to provide the sample in the context of the system_off example; I just haven't gotten to it.  In discussion with @anangl we believe there's nrfx extensions necessary to handle this, and he'll be coordinating that side.

pabigot, 2021-02-02:
#31850 shows how to do RAM retention on a per-application basis.  Infrastructure support requires significantly more effort, even if it's Nordic specific (and it probably should not be).