zephyrproject-rtos/zephyr #31091: usb: usb_transfer_sync deadlocks on disconnect/cancel transfer

xhpohanka, 2021-01-04:
`usb_transfer_sync()` blocks on semaphor that is normally released in a callback

```
static void usb_transfer_sync_cb(uint8_t ep, int size, void *priv)
{
	struct usb_transfer_sync_priv *pdata = priv;

	pdata->tsize = size;
	k_sem_give(&pdata->sem);
}

```

however when the transfer is interrupted (eg. by disconnecting the cable in my test setup with stm32) that callback is never called as we can see in `usb_transfer_work()`

```
static void usb_transfer_work(struct k_work *item)
{
...
done:
...
		/* Transfer completion callback */
		if (trans->status != -ECANCELED) {
			cb(ep, tsize, priv);
		}
	}
}
```

I do not see any obvious fix now, probably we can move the sync semaphore to the `usb_transfer_data` structure and check that case...

jfischer-no, 2021-01-04:
@xhpohanka I guess it happens in your custom USB application? It is in responsibility of your application to cancel an endpoint transfer on USB_DC_DISCONNECTED event, for example https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/bluetooth.c#L223

xhpohanka, 2021-01-04:
If I understand it correctly the bluetooth example does exactly what I'm describing in the first post. It calls an `usb_cancel_transfer()` which sets the transfer status to `-ECANCELED` and thus the callback is never called.

```
void usb_cancel_transfer(uint8_t ep)
{
	struct usb_transfer_data *trans;
	unsigned int key;

	key = irq_lock();

	trans = usb_ep_get_transfer(ep);
	if (!trans) {
		goto done;
	}

	if (trans->status != -EBUSY) {
		goto done;
	}

	trans->status = -ECANCELED;
	k_work_submit_to_queue(&USB_WORK_Q, &trans->work);

done:
	irq_unlock(key);
}
```
However the semaphor used in the `usb_transfer_sync()` is created on the stack and freed only in its never called callback.

```
static void usb_transfer_sync_cb(uint8_t ep, int size, void *priv)
{
	struct usb_transfer_sync_priv *pdata = priv;

	pdata->tsize = size;
	k_sem_give(&pdata->sem);
}

int usb_transfer_sync(uint8_t ep, uint8_t *data, size_t dlen, unsigned int flags)
{
	struct usb_transfer_sync_priv pdata;
	int ret;

	k_sem_init(&pdata.sem, 0, 1);

	ret = usb_transfer(ep, data, dlen, flags, usb_transfer_sync_cb, &pdata);
	if (ret) {
		return ret;
	}

	/* Semaphore will be released by the transfer completion callback */
	k_sem_take(&pdata.sem, K_FOREVER);

	return pdata.tsize;
}
```

jfischer-no, 2021-01-05:
> However the semaphor used in the `usb_transfer_sync()` is created on the stack and freed only in its never called callback.

okay, I missed sync part. That is bad.

> I do not see any obvious fix now, probably we can move the sync semaphore to the `usb_transfer_data` structure and check that case...

I do not see an easier solution as well. `struct k_sem sem;` in `struct usb_transfer_data` should be renamed before.
Do you have a patch or started working on it? Otherwise, I will prepare it today.

xhpohanka, 2021-01-05:
I do not have a patch yet. I do not know if eg. we cannot remove the `(trans->status != -ECANCELED)` check for calling the callback. Maybe it would make better sense to call the callback in any case but with the status as an argument, but it needs an API change...

The most easiest would be probably to add whole `struct usb_transfer_sync_priv` into the `struct usb_transfer_data` and deal with it on unexpected transfer end.

PR #31557 (2021-01-26): usb: fix possible deadlock in usb_transfer_sync()
Syncing here is done with a semaphore declared on a function stack
and released in a callback. In case of a cancelled transfer (eg.
cable disconnect) the callback is never called.

The fix introduces a periodic check if a transfer is still valid.

Fixes #31091
closed 2021-01-26 by nashif