zephyrproject-rtos/zephyr #31100: Recvfrom not returning -1 if UDP and len is too small for packet.

emillindq, 2021-01-05:
**Describe the bug**
According to POSIX, recvfrom should return -1 with errno set to EMSGSIZE if trying to receive an UDP datagram but the provided buffer size is too small for the receiving packet. In zephyr's socket library the packet is truncated and it continues as if it's actually wasn't truncated. Failing silently like this risks to give rise to hard-to-find bugs, at least if using POSIX function recvfrom and expect it to behave POSIXy (which is what happened for me). 

**What have you tried to diagnose or workaround this issue?**
The culprit is https://github.com/zephyrproject-rtos/zephyr/blob/5c7151b02c5567d3897ef28145a27f1dac3612b5/subsys/net/lib/sockets/sockets.c#L984-L987

**To Reproduce**
Steps to reproduce the behavior:
1. Try to receive an UDP packet using recvfrom, and provide a too small buffer.
2. The packet will be trunced and no error given.

**Expected behavior**
That POSIX function recvfrom returns -1 and errno is set to EMSGSIZE.

**Impact**
Not severe, but future users will appreciate the error given.

**Environment (please complete the following information):**
Not applicable

**Additional context**
I was using ethernet on STM32H747I_DISCO_M7

**NOTE**
I can create a merge request implementing said behaviour, but I wanted to hear if this is the consensus first. I'm thinking it was implemented this way for some reason, but I couldn't find anything in the related commit messages or issues.

@jukkar Please comment on this. Thanks!

jukkar, 2021-01-05:
The https://pubs.opengroup.org/onlinepubs/009696699/functions/recvfrom.html does not document `EMSGSIZE` errno, also Linux manual page for recvfrom does not document it. The Linux `recvfrom` manual page says that user can set `MSG_TRUNC` flag, in which case the `recvfrom` will return the real length of the packet or datagram, even when it was longer than the passed buffer.
Where did you find documentation for returning `EMSGSIZE`?

emillindq, 2021-01-06:
> Where did you find documentation for returning `EMSGSIZE`?

After further investigation it seems like the references I had were either not explicitly following POSIX standard, or just forum posts 🙄

All good and thanks for your input @jukkar !

closed 2021-01-06 by emillindq