zephyrproject-rtos/zephyr #31114: Bluetooth: Which coding (S2 vs S8) is used during advertising on Coded PHY?

lukasbrchl, 2021-01-05:
Hi,

Just a quick question. I haven't been able to find out which coding scheme (S2 vs S8) is used during Coded PHY advertisements. I just saw that S2 or S8 could be set during connection. Could you please clarify which one is used during the advertisements?

Thanks.

cvinayak, 2021-01-09:
Bluetooth Specification v5.2, Vol 4, Part E, Section 7.8.53:
> If advertising on the LE Coded PHY, the S=8 coding shall be assumed.

joerchan, 2021-01-11:
> If advertising on the LE Coded PHY, the S=8 coding shall be assumed.

This seems like it is related to the maximum packet length, and interval, i.e you cannot set a packet that could not be set in the configured advertising interval and vice versa.

Assuming that S8 is used seems to be  the most reasonable, but selecting coding scheme is up to the controller, and the Host has no way according to HCI to specify preferred coding.

lukasbrchl, 2021-01-12:
Hello guys, I apologize for being relentless, but I didn't understand the points you mentioned. Could you please address it a little more? Does the Zephyr BT Controller use coding scheme S8 for advertising on Coded PHY or not?

joerchan, 2021-01-12:
@lukasbrchl The zephyr controller at the moment uses S8 coding.

lukasbrchl, 2021-01-12:
Great, thanks for the reply!

closed 2021-01-12 by lukasbrchl