zephyrproject-rtos/zephyr #31134: LittleFS: Error Resizing the External QSPI NOR Flash in nRF52840dk

DineshDK03, 2021-01-06:
**Describe the bug**
I tried running the [USB mass storage](https://docs.zephyrproject.org/latest/samples/subsys/usb/mass/README.html#usb-mass-storage-sample-application) sample in that I can't able to use the whole 64MiBi of external qspi_nor flash in nRF52840dk  for littleFS partition. I tried changing the reg value in the partition table as <0x0  0x4000000>  inside the overlay file Also I tried changing the label as label = "storage_lfs" and replaced the label name accordingly in the source code as well. 

**To Reproduce**
Steps to reproduce the behavior:
1. Edit nrf52840dk_nrf52840.overlay file's Line 16 into this `reg = <0x00000000 0x4000000>;` to get 64Mb size
2. Follow this [link](https://docs.zephyrproject.org/latest/samples/subsys/usb/mass/README.html#littlefs-example) for  build, flash, and mount as LittleFS
3. During the process of LFS mounting in the Host system, make sure changed the `block_count=16384`

**Expected behavior**
It is supposed to access 64Mb size and able to use  basic Linux file system operations like mkdir, cp, mv, etc.,

**Logs and Console Outputs**
minicom terminal output is as follows,
```
*** Booting Zephyr OS build zephyr-v2.4.0-2674-g4ca7942411f7  ***
Area 4 at 0x0 on MX25R64 for 67108864 bytes
[00:00:00.368,164] <inf> littlefs: LittleFS version 2.2, disk version 2.0
[00:00:00.370,300] <inf> littlefs: FS at MX25R64:0x0 is 16384 0x1000-byte blocks with 512 cycle
[00:00:00.370,300] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.370,697] <inf> littlefs: /lfs mounted
Mount /lfs: 0
/lfs: bsize = 16 ; frsize = 4096 ; blocks = 16384 ; bfree = 16382
/lfs opendir: 0
End of files
[00:00:00.431,610] <inf> main: The device is put in USB mass storage mode.
```
but while running system log to determine device name it shows only 131kB only available as follows,
```
Jan 06 11:49:56 arch kernel: scsi 9:0:0:0: Direct-Access    ZEPHYR  ZEPHYR USB DISK 0.01 PQ: 0 ANSI: 0 CCS 
Jan 06 11:49:56 arch kernel: sd 9:0:0:0: [sdg] 256 512-byte logical blocks: (131 kB/128 KiB) 
Jan 06 11:49:56 arch kernel: sd 9:0:0:0: [sdg] Write Protect is off 
Jan 06 11:49:56 arch kernel: sd 9:0:0:0: [sdg] Mode Sense: 03 00 00 00 
Jan 06 11:49:56 arch kernel: sd 9:0:0:0: [sdg] No Caching mode page found 
Jan 06 11:49:56 arch kernel: sd 9:0:0:0: [sdg] Assuming drive cache: write through 
Jan 06 11:49:57 arch kernel: sd 9:0:0:0: [sdg] Attached SCSI removable disk
```
Also during mounting the littleFS partition with the host system by following this instruction in this link. I changed the `block_count =16384`  mounting process done successfully but most linux commands like mkdir, cp commands are not working(`mkdir: cannot create directory 'new': Invalid argument`). But at the same time during the littlefs mounting process changing the `block_count = 32`  (example's default value) and doing so makes linux commands work properly. 

**Environment (please complete the following information):**
 - OS: Linux (Arch Linux to be specific)

jfischer-no, 2021-01-06:
You need to adjust [DISK_VOLUME_SIZE](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/usb/mass/Kconfig#L77) accordingly. The default value is 0x20000 = `256 512-byte logical blocks`.

DineshDK03, 2021-01-07:
Thank you @jfischer-no, it worked

closed 2021-01-07 by DineshDK03