zephyrproject-rtos/zephyr #31143: samples: audio: sof: compilation issue, include file not found.

ABOSTM, 2021-01-07:
**Describe the bug**
While running automatic tests on STM32 test bench,
samples/audio/sof build fails to build.
This appends on all boards (STM32 boards).

**To Reproduce**
Steps to reproduce the behavior:
1. scripts/twister --ninja --runtime-artifact-cleanup -b  -p nucleo_f429zi  -T samples/audio/sof/ --outdir ./log_db_audio
2. See error

**Expected behavior**
Build successful or sample not build for those boards

**Logs and console output**
```
In file included from /home/osboxes/zephyr/zephyr_project/modules/audio/sof/zephyr/../src/arch/xtensa/include/arch/drivers/timer.h:13,
                 from /home/osboxes/zephyr/zephyr_project/modules/audio/sof/zephyr/../src/include/sof/drivers/timer.h:11,
                 from /home/osboxes/zephyr/zephyr_project/modules/audio/sof/src/lib/clk.c:10:
/home/osboxes/zephyr/zephyr_project/modules/audio/sof/zephyr/../src/include/sof/drivers/interrupt.h:11:10: fatal error: platform/drivers/interrupt.h: No such file or directory
   11 | #include <platform/drivers/interrupt.h>
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain Zephyr SDK
 - Commit SHA: 536f0f5acbce4c974709d5c5ce224eaac6b08022

**Additional context**
In `sample.yaml` there is:
```
    integration_platforms:
      - intel_adsp_cavs15
```
So is it normal that twister try to build for other platforms ?
(I don't catch exactly the difference between `integration_platforms `and `platform_allow`)

ABOSTM, 2021-01-07:
^^ @nashif @andyross

lgirdwood, 2021-01-07:
@nashif are you ok to fix this and #31153 to temp exclude the non CAVS SoC until @lyakh generic work is ready.

galak, 2021-01-07:
@lgirdwood I think something like:

```
diff --git a/modules/Kconfig.sof b/modules/Kconfig.sof
index 17e40df34b..fb5b5c125e 100644
--- a/modules/Kconfig.sof
+++ b/modules/Kconfig.sof
@@ -3,5 +3,6 @@
 
 config SOF
        bool "Sound Open Firmware (SOF)"
+       depends on SOC_FAMILY_INTEL_ADSP
        help
          Build Sound Open Firmware (SOF) support.
diff --git a/samples/audio/sof/sample.yaml b/samples/audio/sof/sample.yaml
index 32ba0df254..c726df2fc4 100644
--- a/samples/audio/sof/sample.yaml
+++ b/samples/audio/sof/sample.yaml
@@ -7,5 +7,6 @@ tests:
   sample.audio.sof:
     tags: sof
     build_only: true
+    filter: ( CONFIG_SOF )
     integration_platforms:
       - intel_adsp_cavs15
```

as a start.

galak, 2021-01-07:
However even when I do that, I get the following on intel_adsp_cavs15:

```
In file included from /home/galak/git/modules/audio/sof/src/platform/intel/cavs/platform.c:46:
/home/galak/git/modules/audio/sof/zephyr/../src/include/kernel/ext_manifest.h:32:10: fatal error: rimage/sof/kernel/ext_manifest.h: No such file or directory
   32 | #include <rimage/sof/kernel/ext_manifest.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

galak, 2021-01-07:
Does this require SDK 0.12.0?

lgirdwood, 2021-01-07:
@lyakh any comment here, IIRC you have some PR pending with rimage updates for manifest (like above error) ?

nashif, 2021-01-07:
you need to run `west sof update`

nashif, 2021-01-07:
> depends on SOC_FAMILY_INTEL_ADSP

we should not add such dependency, the idea is to make SOF also work on non Intel SoCs...

lets just limit what platforms this can run on using explicit platform filters.

galak, 2021-01-07:
> you need to run `west sof update`

???? what is this doing?  Why isn't this just part of the zephyr west.yml?

nashif, 2021-01-07:
> ???? what is this doing? Why isn't this just part of the zephyr west.yml?

This is fetching git submodules needed by SOF. Tried putting those in west, but then the submodules have their own submodules....

nashif, 2021-01-07:
> This is fetching git submodules needed by SOF. Tried putting those in west, but then the submodules have their own submodules....

same story as with espressif module with a PR pending, we really need the west support for submodules to get over this type of issues...

galak, 2021-01-07:
So I feel like this needs more discussion as this seems to violates having modules being "copied" into zephyr github.

galak, 2021-01-07:
@carlescufi / @ioannisg -- I think we need to discuss how zephyr as a project will handle modules w/git submodules as a general rule.

nashif, 2021-01-07:
> So I feel like this needs more discussion as this seems to violates having modules being "copied" into zephyr github.

what is being copied in zephyr github? The modules are not in Zephyr's github.

galak, 2021-01-07:
> > 
> 
> we should not add such dependency, the idea is to make SOF also work on non Intel SoCs...
> 
> lets just limit what platforms this can run on using explicit platform filters.

We'll the SOF code current seems limited to xtensa and maybe just the Intel SoCs at this point.  So I agree in principle this shouldn't be the case but if I enable CONFIG_SOF and another platform it's looking for xtensa code.  So feels like for right now CONFIG_SOF is only viable on the Intel xtensa SoCs.  When that changes we can remove the `depends on` that I suggest.

nashif, 2021-01-08:
> So I agree in principle this shouldn't be the case but if I enable CONFIG_SOF and another platform it's looking for xtensa code.

NXP Platforms with audio DSP also can use SOF on top of Xtensa.

PR #31171 (2021-01-08): Do not use west extension with SOF
Fixes #31143
Fixes #31153
closed 2021-01-08 by nashif