zephyrproject-rtos/zephyr #31145: Litex-vexriscv address misaligned with dumb_http_server example

pawelsag, 2021-01-07:
**Bug description**
On Arty-7 board when running the eth dumb_http_server with vexriscv core
I encountered the error with misaligned memory access.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=litex_vexriscv
3. make
The bitstream was generated with https://github.com/litex-hub/zephyr-on-litex-vexriscv

**Expected behavior**
After sending the HTTP request using curl I expected to see the returned page.
I have configured the network and run the following command  `curl http://192.168.2.4:8080`

**Logs and console output**
On arty board the following error occurs:
```
[00:00:08.810,000] <err> os: Exception cause Load address misaligned (4)
[00:00:08.810,000] <err> os: Faulting instruction address = 0x40010dec
[00:00:08.810,000] <err> os:   ra: 0x40010ddc  gp: 0xffffffff  tp: 0xffffffff  t0: 0x00000080
[00:00:08.810,000] <err> os:   t1: 0x4001c7c0  t2: 0xfffffff5  t3: 0xffffffff  t4: 0xffffffff
[00:00:08.810,000] <err> os:   t5: 0xffffffff  t6: 0xffffffff  a0: 0x40023392  a1: 0x00000000
[00:00:08.810,000] <err> os:   a2: 0x40022ed8  a3: 0x40023392  a4: 0x00000001  a5: 0x0000003c
[00:00:08.810,000] <err> os:   a6: 0x00000014  a7: 0x0000003c

[00:00:08.810,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:08.810,000] <err> os: Current thread: 0x4001c7c0 (unknown)
[00:00:08.810,000] <err> os: Halting system

```
The faulting instruction points to https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp2.c#L1269

The asm snippet code:
```
size_t tcp_options_len = th ? (th->th_off - 5) * 4 : 0;
40010de4:	00000993          	li	s3,0
40010de8:	fa0500e3          	beqz	a0,40010d88 <tcp_in+0x50>
40010dec:	00c52983          	lw	s3,12(a0)

```
As we can see we are loading the 32 bit from a misaligned address.
The value in `a0` register equals to `0x40023392` that can't be divided by 4.

The problem occurs since this commit: 70dae094ba2dd3e1f02684a8d673d434966b8b76
The commit introduces tcp_v2 implementation.
As we can see the problem is caused by `th_off` filed here: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp2_priv.h#L126
The compiler generates the load word instruction instead of the load byte instruction.
I have tested it with other versions of gcc-riscv: 8.2.0 and clang-riscv from trunk.
And I'm quite sure this is correct behavior, cause the struct filed offset is aligned to 4 bytes.
The problem is that somewhere in the code the unaligned memory is cast to this struct.

**Environment:**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.4
 - Commit: 1c493b4a7706e9934949f8d23f8d8eddc7256200

**Additional context**
When changing `tx_off` to just `uint8_t` and getting rid of the bitfierlds the compiler generates:
```
40010dec:	00c52983          	lbu	s3,12(a0)
```
So this is one of the possible solutions the second is to assure that the memory is correctly aligned.

jukkar, 2021-01-07:
Hi @pawelsag, I have a fix proposal at #31159, could you try if it fixes the issue and report back?

pawelsag, 2021-01-07:
Hi @jukkar your fix works for me, thanks

PR #31159 (2021-01-08): net: tcp2: Mark tcphdr struct as packed
As the tcp header struct can be cast to unaligned memory, mark
it as packed and access fields using UNALIGNED_GET/PUT when
needed.

Fixes #31145
commit d904f955 (2021-01-08): net: tcp2: Mark tcphdr struct as packed

As the tcp header struct can be cast to unaligned memory, mark
it as packed and access fields using UNALIGNED_GET/PUT when
needed.

Fixes #31145
closed 2021-01-08 by jukkar