zephyrproject-rtos/zephyr #31150: tests/subsys/canbus/isotp/conformance: failing on nucleo_f746zg

erwango, 2021-01-07:
**Describe the bug**
tests/subsys/canbus/isotp/conformance: failing on nucleo_f746zg (and stm32f3_disco)

**To Reproduce**
1- twister --hardware-map ../map.yaml --device-testing -p nucleo_746zg -T tests/subsys/canbus/isotp/conformance
2- See error

**Expected behavior**
Test should pass

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
$ cat /local/mcu/zephyrproject/zephyr/twister-out/nucleo_f746zg/tests/subsys/canbus/isotp/conformance/can.isotp.conformance/handler.log
```
I: Init of CAN_1 done
*** Booting Zephyr OS build zephyr-v2.4.0-2807-gb34d05592666  ***
Running test suite isotp_conformance
===================================================================
START - test_send_sf
PASS - test_send_sf
===================================================================
START - test_receive_sf
I: SF DL does not match. Ignore
I: SF DL does not match. Ignore
PASS - test_receive_sf
===================================================================
START - test_send_sf_ext
PASS - test_send_sf_ext
===================================================================
START - test_receive_sf_ext
I: SF DL does not match. Ignore
I: SF DL does not match. Ignore
PASS - test_receive_sf_ext
===================================================================
START - test_send_data
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
desired bytes:
2c 90 b6 e7 45 83 8d 71 
received (8 bytes):
26 0f 1e fa 17 cb 63 b9 
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:253: check_frame_series: (ret not equal to 0)
Data differ
FAIL - test_send_data
===================================================================
START - test_send_data_blocks
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
E: Msgq 0x2001029c overflowed. Frame ID: 0x10
desired bytes:
26 36 7d 91 d7 f5 c5 81 
received (8 bytes):
26 0f 1e fa 17 cb 63 b9 
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:253: check_frame_series: (ret not equal to 0)
Data differ
FAIL - test_send_data_blocks
===================================================================
START - test_receive_data
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:249: check_frame_series: (frame.dlc not equal to desired->length)
DLC of frame nr 0 differ. Desired: 3, Got: 8
FAIL - test_receive_data
===================================================================
START - test_receive_data_blocks
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x8004c55
E: r0/a1:  0x08004c55  r1/a2:  0x00000000  r2/a3:  0x20010b88
E: r3/a4:  0x20010680 r12/ip:  0x150b0000 r14/lr:  0x080067ad
E:  xpsr:  0x21000024
E: Faulting instruction address (r15/pc): 0x080088d4
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20010680 (idle 00)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.12
 - Commit SHA: zephyr-v2.4.0-2807-gb34d055926

**Additional context**
No additional wire used with the board

martinjaeger, 2021-01-07:
@erwango do you have a CAN transceiver added to the board?

erwango, 2021-01-07:
> @erwango do you have a CAN transceiver added to the board?

@martinjaeger I confirm there is no transceiver.

martinjaeger, 2021-01-07:
Turns out that I was not aware that the loopback mode feeds back the tx signal to rx internally. So it does not require a transceiver and the test should work without external hardware. Sorry for the confusion.

FWIW: This test passes on my custom hardware based on STM32G431 (with the new WIP CAN-FD drivers). But I can also confirm it fails in the same way as described above on the `nucleo_l452re` (don't have a `nucleo_f746`). So it might be a driver issue and not ISO-TP. Certainly needs further investigation.

martinjaeger, 2021-01-08:
`git bisect` shows that there are two separate issues that make this test fail:

- A failure because of no free filters. This was introduced in 8b6c1bd4b7a0ae4e1d1b3819d5db728120218c6b and will be fixed by #31161
- The MPU fault above, which was introduced in 05275ecf6e2d63e19953a7ebb3bae99b7cf13487.

carlescufi, 2021-01-26:
Single platform and subsystem, downgrading to low.

closed 2021-04-11 by github-actions[bot]