zephyrproject-rtos/zephyr #31153: twister build of samples/audio/sof/sample.audio.sof fails on most platforms

galak, 2021-01-07:
`./scripts/twister -T samples/audio/sof/ -l`

Gets a slew of errors like:

```
ERROR   - m2gl025_miv               samples/audio/sof/sample.audio.sof                 FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/m2gl025_miv/samples/audio/sof/sample.audio.sof/build.log
INFO    - Total complete:    5/ 308   1%  skipped:   12, failed:    5ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/audio/sof for minnowboard

ERROR   - minnowboard               samples/audio/sof/sample.audio.sof                 FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/minnowboard/samples/audio/sof/sample.audio.sof/build.log
INFO    - Total complete:    6/ 308   1%  skipped:   12, failed:    6ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/audio/sof for qemu_x86_64

ERROR   - qemu_x86_64               samples/audio/sof/sample.audio.sof                 FAILED: Cmake build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_x86_64/samples/audio/sof/sample.audio.sof/build.log
INFO    - Total complete:    7/ 308   2%  skipped:   12, failed:    7ERROR   - Cmake build failure: /home/galak/git/zephyr/samples/audio/sof for litex_vexriscv

```

galak, 2021-01-07:
@nashif fyi.

erwango, 2021-01-07:
Same as https://github.com/zephyrproject-rtos/zephyr/issues/31143

lgirdwood, 2021-01-07:
@nashif @galak we probably need a ```ifdef INTEL_CAVS_FAMILY``` around SOF atm as today it only builds against certain xtensa SoCs. 
Fwiw, @lyakh is doing some updates to make the code fully generic in time for SOF 1.7 and Zephyr 2.5 (there are some drivers/arch code that is getting pulled to the SOF build in some places).

PR #31171 (2021-01-08): Do not use west extension with SOF
Fixes #31143
Fixes #31153
closed 2021-01-08 by nashif