zephyrproject-rtos/zephyr #31162: Mapping between existing and new system power management states

pabigot, 2021-01-07:
There's ongoing effort to improve Zephyr's power management solution, which includes switching to a [standard set of states](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L23) based on ACPI/Linux, to replace the [current sleep/deep-sleep trinary solution](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power.h#L25) that has no clear guidance for what the states imply (other than deep sleep means state is lost).

To that end we need to determine the state mapping for any SoC that currently supports the existing power states.  This issue is a place for all SoC maintainers and interested parties to propose mappings and develop a common understanding of what the new states mean and how they'll be used.

Note these basic assumptions:
* [PM_STATE_ACTIVE](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L26) means the CPU is running normally.
* [PM_STATE_RUNTIME_IDLE](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L34) corresponds to what Zephyr normally does when the idle thread is the only active thread.  E.g. for ARM this is blocking in `WFE`/`WFI` for some interrupt to occur.

In Linux terms these are both runtime states, and neither is under control of Zephyr's system power management infrastructure.

The real system states used when power management is enabled are below, and existing uses of `POWER_STATE{,_DEEP}_SLEEP_{1,2,3}` should map to one of these states:
* [PM_STATE_SUSPEND_TO_IDLE](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L45)
* [PM_STATE_STANDBY](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L57)
* [PM_STATE_SUSPEND_TO_RAM](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L69)
* [PM_STATE_SUSPEND_TO_DISK](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L81)
* [PM_STATE_SOFT_OFF](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L94)

The map does not need to be consistent for all SoCs; e.g. that vendor A's `POWER_STATE_SLEEP_1` corresponds to `PM_STATE_SUSPEND_TO_IDLE` while vendor B maps it to `PM_STATE_STANDBY` is not a concern.  We just want the existing supported states to use the new state names.  This will enable @ceolin and others to prepare commits that move us toward the new states.

Discussion that led to the selection of these states can be found in #24228, #27315, #29505.  Where there is ambiguity the references in https://github.com/zephyrproject-rtos/zephyr/issues/24228#issuecomment-663584642 may be useful to determine intended behavior.

pabigot, 2021-01-07:
For Nordic boards I believe all existing states would map to [PM_STATE_SOFT_OFF](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L94)  as the only non-runtime sleep mode on most if not all Nordic devices is a system-off where some RAM data may be retained but the system shut down and wake only on an external signal from a GPIO or other peripheral (not including a timer).

[PM_STATE_SUSPEND_TO_RAM](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L69) could be supported if/when #31087 is addressed and infrastructure is added to detect the wakeup cause and restore the application and its peripherals to an active state that includes state inherited from the previous session.

erwango, 2021-01-07:
^^@erwango

erwango, 2021-01-08:
Here is STM32 mapping (based on STM32L4/STM3WB):
- Stop 0, Stop 1, Stop 2 > [PM_STATE_SUSPEND_TO_IDLE](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L46)
- Standby with SRAM2 > [PM_STATE_SUSPEND_TO_RAM](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L80) (requires RAM retention support)
- Standby and Shutdown > [PM_STATE_SOFT_OFF](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L95)

Tbh, I'm not sure clean on the frontier between [PM_STATE_SUSPEND_TO_IDLE](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L46) and  [PM_STATE_STANDBY](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L56) so Stop 2 could be [PM_STATE_STANDBY](https://github.com/zephyrproject-rtos/zephyr/blob/4ba387829b9febd60dc97d4db39d833b8aa56058/include/power/power_state.h#L56).

Low Power Run and Low Power Sleep modes could not be matched with the proposed states. They are similar to PM_STATE_ACTIVE and PM_STATE_RUNTIME_IDLE with a clock frequency reduced to 2MHz.

One question: how can wen handle multiple states mapping to a single PM state (injective mapping) ?
Today we have the flexibility of mapping to POWER_STATE_SLEEP_X, but IIUC we lose this flexibility with the new PM states.

pabigot, 2021-01-09:
The existing SiLabs mapping has:

* POWER_STATE_SLEEP_1: EM1 Sleep
* POWER_STATE_SLEEP_2: EM2 Deep Sleep
* POWER_STATE_SLEEP_3: EM3 Stop

Reviewing the semantics of these this doesn't seem quite ideal.  The best mapping I can come up informed by [this overview](https://www.silabs.com/products/mcu/32-bit/efm32-energy-modes) and the EFR32xG12 reference manual is:

* PM_STATE_ACTIVE: EM0
* PM_STATE_RUNTIME_IDLE: EM1
* PM_STATE_SUSPEND_TO_IDLE: Either EM2 or EM3.  The SiLabs functional distinction does not appear to match well to ACPI state distinction.
* PM_STATE_STANDBY: Same as PM_STATE_SUSPEND_TO_IDLE
* PM_STATE_SUSPEND_TO_RAM: EM4 Hibernate
* PM_STATE_SUSPEND_TO_DISK: N/A
* PM_STATE_SOFT_OFF: EM4 Shutoff

It appears waking due to elapsed time (not Zephyr system clock) is possible from all of these states.  I suspect full capability will require API specific to a SiLabs custom power management policy that does not distinguish between PM_STATE_SUSPEND_TO_IDLE and PM_STATE_STANDBY, and instead determines which of EM2 and EM3 should be entered based on constraints provided by the application.  This *could* be handled by treating PM_STATE_SUSPEND_TO_IDLE as EM2 and PM_STATE_STANDBY as EM3 but that is not justified 

I haven't used these chips since they were Energy Micro, so @chrta or @mnkp may be able to provide more up-to-date insight.

pabigot, 2021-01-09:
> One question: how can wen handle multiple states mapping to a single PM state (injective mapping) ?
> Today we have the flexibility of mapping to POWER_STATE_SLEEP_X, but IIUC we lose this flexibility with the new PM states.

I think we should plan to have SOC-specific default power management policies, that support API extensions to allow applications to control which specific mode is used to implement a Zephyr standard mode.  I don't know yet what the rest of the power management API will look like so can't propose how the extensions would work.

ceolin, 2021-01-09:
> Here is STM32 mapping (based on STM32L4/STM3WB):
>
> * Stop 0, Stop 1, Stop 2 > [...]

Currently is not flexible to be fair, we have a fixed number of `POWER_STATE_SLEEP_`. It just happen that the number of these states is enough for this platform, but it could not be the case. But yeah, that is something we need to think about, when ACPI-like states don't map well. Let me think over the weekend about it.

pabigot, 2021-01-10:
>  It just happen that the number of these states is enough for this platform, but it could not be the case.

We need to be careful here.  The state names have meanings, and are intended to map to ACPI semantics.

If some SoC has three power levels that all correspond in basic behavior to `PM_STATE_STANDBY`, we **must not** arbitrarily assign them to `IDLE` or `SUSPEND` states just because those enumerations are not being used for their documented purpose.  That would be worse than providing 3 arbitrary versions of SLEEP, because at least there we "knew" they all did the same basic thing, but with different side effects.

I really think this sort of distinction needs to be SOC-specific internal substates of the standard Zephyr states, managed by SoC-specific policy implementations.  When using a Zephyr-wide default/generic policy `PM_STATE_STANDBY` would be allowed, and you'd get whatever single state the SoC maintainer best matches that, and `IDLE` and `SUSPEND` states would not be supported.

chrta, 2021-01-11:
Regarding Silabs EXX32:

I am not experienced with the low-power modes, but i think the overview https://github.com/zephyrproject-rtos/zephyr/issues/31162#issuecomment-757337571 is correct. The main differences between EM2 and EM3 is which peripherals are still active and therefore which events trigger a wake-up.
I also think that the decision if EM2 or EM3 should be used, must be up to the application.

Is there documentation how wake-up sources should be configured in zephyr's power management solution? I mean should there be a timer running and/or should wake-up work if something happens on a peripheral e.g. UART? For EXX32 this would mean that the peripheral drivers need to be adapted for this.

noelleclement, 2021-01-11:
> Here is STM32 mapping (based on STM32L4/STM3WB):
> 
> - Stop 0, Stop 1, Stop 2 > PM_STATE_SUSPEND_TO_IDLE
> - Standby with SRAM2 > PM_STATE_SUSPEND_TO_RAM (requires RAM retention support)
> - Standby and Shutdown > PM_STATE_SOFT_OFF

Is there overall power state description/mapping for STM32, so this could be applied to all STM32 SoCs?  (e.g. STM32L1x only has one stop mode, which also would map to PM_STATE_SUSPEND_TO_IDLE)

ceolin, 2021-01-11:
> > It just happen that the number of these states is enough for this platform, but it could not be the case.
> 
> We need to be careful here. The state names have meanings, and are intended to map to ACPI semantics.

Yes, completely agree. 
> If some SoC has three power levels that all correspond in basic behavior to `PM_STATE_STANDBY`, we **must not** arbitrarily assign them to `IDLE` or `SUSPEND` states just because those enumerations [...]

Didn't say to do that. The dilemma is, `STATE_SLEEP_1`, `STATE_SLEEP_2`, ... were meaningless from the application perspective. I mean, one could not infer what was the difference between them, and different platforms could do different things. in other hand when we move to ACPI-like states, an application know  exactly what a power state is supposed to do, but we lose flexibility on the platform implementation.

> I really think this sort of distinction needs to be SOC-specific internal substates of the standard Zephyr states, managed by SoC-specific policy implementations. When using a Zephyr-wide [...]

nashif, 2021-01-12:
good reference for STM:

[image]

https://www.st.com/content/ccc/resource/training/technical/product_training/ce/57/a3/86/7a/3d/4d/87/STM32L4_System_Power.pdf/files/STM32L4_System_Power.pdf/jcr:content/translations/en.STM32L4_System_Power.pdf

raveslave, 2021-01-18:
don't forget a state for DEAD_BATT situations as this usually requires some specials for bringing up the system, i.e. could be Type-C PD negotiation or early boot-sequence USB-enumeration to happen before you can enter PM_STATE_ACTIVE

pabigot, 2021-01-18:
> don't forget a state for DEAD_BATT situations as this usually requires some specials for bringing up the system

Does that deserve a specific PM state?  It's not something the generic power management infrastructure would be aware of.  SoC-specific startup has to detect conditions on startup and ensure that the system is brought up in an active mode, which might be a degraded substate of `PM_STATE_ACTIVE`.

raveslave, 2021-01-18:
I saw it more as a quite common entry-point where you'd want to stay until x-number of conditions are met to continue. Having it part of the PM would be nice for structure, logging options but also in multi-core systems where one core or physical mcu might want to ask for the PM state

pabigot, 2021-01-21:
@galak Added to dev-revew as a better anchor issue for discussion of https://github.com/zephyrproject-rtos/zephyr/pull/31273#issuecomment-761086654, which is why that PR is in dev-review.

For context, my position remains that we need to support MCUs that have multiple power configurations at runtime, not just when idle.  This follows from https://github.com/zephyrproject-rtos/zephyr/issues/31162#issuecomment-758631584 where we see STM32 has three active states (80 MHz, 26 MHz, 2 MHz), that have two runtime idle states, none of which are idle states from the perspective of Zephyr.

I believe the power management infrastructure is ideally positioned to support transition between the variant active states, and consequently devicetree description of power states should not be limited to those that happen to be what Zephyr (or Linux) calls idle states.

gmarull, 2021-07-14:
@nashif @ceolin shouldn't that issue be closed? I think the main issue described here has been addressed.

ceolin, 2021-07-14:
@gmarull yep.  I don't see any reason to keep it open, we can always reference it even after closed.

closed 2021-07-14 by ceolin