zephyrproject-rtos/zephyr #31164: Problem to build lorawan from samples

GABS9718, 2021-01-07:
**Describe the bug**
Hi, I have been trying to set up the LoRa with the nRF52840, but I'm can't build the file.

**To Reproduce**
Steps to reproduce the behavior:

Add the line lora0 = &lora; in the aliases options from the .dts related to the board nrf52840dk_nrf52840
Add
_lora: sx1276@0 {
  status = "okay";
  compatible = "semtech,sx1276";
  reg = <0>;
  label = "sx1276";
  reset-gpios = <&gpio0 13 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
  dio-gpios = <&gpio0 1 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
  <&gpio0 2 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
  <&gpio0 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
  <&gpio0 4 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
  <&gpio0 5 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
  spi-max-frequency = <1000000>;
  power-amplifier-output = "pa-boost";
};_
to the spi1 options in the same file.
Change the prf.conf from samples/lorawan/class_a
_CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n

CONFIG_LOG=y
CONFIG_SPI=y
CONFIG_SPI_1=y
CONFIG_SPI_NRFX=y
CONFIG_GPIO=y
CONFIG_PRINTK=y

CONFIG_LORA=y
CONFIG_LORA_SHELL=y
CONFIG_LORA_SX1276=y
CONFIG_HAS_SEMTECH_RADIO_DRIVERS=y
CONFIG_HAS_SEMTECH_LORAMAC=y_

**Expected behavior**
Build the project

**Impact**
I'm getting this error: 
[145/150] Linking C executable zephyr\zephyr_prebuilt.elf 
FAILED: zephyr/zephyr_prebuilt.elf 

**Environment (please complete the following information):**

OS: Windows
Toolchain: GNU ARM Embedded
The newest one from the repository

**Additional context**
I can build other samples, like hello_world and blinky.

LOG:
> -- west build: generating a build system
Including boilerplate (Zephyr base): C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a
-- Zephyr version: 2.4.99 (C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr)
-- Found Python3: C:/Python39/python.exe (found suitable exact version "3.9.0") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/gnu_arm_embedded)
-- Found BOARD.dts: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/include/generated/devicetree_unfixed.h

> warning: LORA_SHELL (defined at drivers/lora/Kconfig:21) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: SHELL (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_LORA_SHELL.html and/or look up
LORA_SHELL in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.
warning: The choice symbol LORA_SX1276 (defined at drivers/lora/Kconfig.sx12xx:31) was selected (set
=y), but no symbol ended up as the choice selection. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_LORA_SX1276.html and/or look up
LORA_SX1276 in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.
Parsing C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration 'C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/prj.conf'
Configuration saved to 'C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/.config'
Kconfig header saved to 'C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/gnu_arm_embedded/bin/arm-none-eabi-gcc.exe
CMake Deprecation Warning at C:/Users/Gabriel/Desktop/nbps/zephyrproject/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

>-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build
-- west build: building application
[145/150] Linking C executable zephyr\zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf 
cmd.exe /C "cd . && C:\gnu_arm_embedded\bin\arm-none-eabi-gcc.exe   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr\zephyr_prebuilt.elf  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/mpu/libarch__arm__core__aarch32__cortex_m__mpu.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/spi/libdrivers__spi.a  zephyr/drivers/lora/libloramac-node.a  zephyr/drivers/serial/libdrivers__serial.a  modules/nordic/lib..__modules__hal__nordic.a  
-Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/thumb/v7e-m/nofp"  -LC:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr  -lgcc  -Wl,--print-memory-usage  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lm  -lc  -lgcc  -L"C:/gnu_arm_embedded/arm-none-eabi"/lib/thumb/v7e-m/nofp  -specs=nano.specs && cd ."
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: app/libapp.a(main.c.obj): in function `main':
C:\Users\Gabriel\Desktop\nbps\zephyrproject\zephyr\samples\lorawan\class_a\build/../src/main.c:50: undefined reference to `lorawan_start'
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Gabriel\Desktop\nbps\zephyrproject\zephyr\samples\lorawan\class_a\build/../src/main.c:63: undefined reference to `lorawan_join'
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Gabriel\Desktop\nbps\zephyrproject\zephyr\samples\lorawan\class_a\build/../src/main.c:71: undefined reference to `lorawan_send'
Memory region         Used Size  Region Size  %age Used
           FLASH:       32636 B         1 MB      3.11%
            SRAM:        9248 B       256 KB      3.53%
        IDT_LIST:         120 B         2 KB      5.86%
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files (x86)\CMake\bin\cmake.EXE' --build 'C:\Users\Gabriel\Desktop\nbps\zephyrproject\zephyr\samples\lorawan\class_a\build'

stoberblog, 2021-01-08:
Looks like the lorawan module is not being enabled in your prj.conf file. Try adding the below to your prj.conf (you may need to change the region you use, maybe use menuconfig to determine the one you want).
CONFIG_LORAWAN=y
CONFIG_LORAMAC_REGION_AS923=y

GABS9718, 2021-01-08:
I have added what u said, but it still doesn't work. My prj.conf file is below

CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n

CONFIG_LOG=y
CONFIG_SPI=y
CONFIG_SPI_0=y
CONFIG_SPI_NRFX=y
CONFIG_GPIO=y
CONFIG_PRINTK=y

CONFIG_LORA=y
CONFIG_LORAWAN=y
CONFIG_LORAMAC_REGION_US915=y
CONFIG_LORA_SHELL=y
CONFIG_LORA_SX1276=y
CONFIG_HAS_SEMTECH_RADIO_DRIVERS=y
CONFIG_HAS_SEMTECH_LORAMAC=y

sophiekovalevsky, 2021-01-08:
@GABS9718 I guess you can not define symbol `CONFIG_LORA_SX1276` cause that's selected by corresponding device tree overlay, instead, try to add `CONFIG_LORA_SX12XX` removing  `CONFIG_LORA_SX1276=y` in your `prj.conf` and see if that works.

GABS9718, 2021-01-08:
I tried and the result is the same unfortunately

sophiekovalevsky, 2021-01-08:
@GABS9718 what's the output that west is throwing this time?

GABS9718, 2021-01-08:
I guess the same:

`-- west build: generating a build system
Including boilerplate (Zephyr base): C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a
-- Zephyr version: 2.4.99 (C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr)
-- Found Python3: C:/Python39/python.exe (found suitable exact version "3.9.0") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/gnu_arm_embedded)
-- Found BOARD.dts: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/include/generated/devicetree_unfixed.h

warning: LORA_SHELL (defined at drivers/lora/Kconfig:21) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: SHELL (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_LORA_SHELL.html and/or look up
LORA_SHELL in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: LORA_SX12XX (defined at drivers/lora/Kconfig.sx12xx:11) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: SPI (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_LORA_SX12XX.html and/or look up
LORA_SX12XX in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: SPI_0 (defined at drivers/spi/Kconfig:39) was assigned the value 'y' but got the value 'n'.
Check these unsatisfied dependencies: SPI (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_SPI_0.html and/or look up SPI_0 in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

Parsing C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration 'C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/prj.conf'
Configuration saved to 'C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/.config'
Kconfig header saved to 'C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/gnu_arm_embedded/bin/arm-none-eabi-gcc.exe
CMake Deprecation Warning at C:/Users/Gabriel/Desktop/nbps/zephyrproject/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build
-- west build: building application
[143/148] Linking C executable zephyr\zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf 
cmd.exe /C "cd . && C:\gnu_arm_embedded\bin\arm-none-eabi-gcc.exe   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr\zephyr_prebuilt.elf  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=C:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/mpu/libarch__arm__core__aarch32__cortex_m__mpu.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/subsys/lorawan/libloramac-node.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  modules/nordic/lib..__modules__hal__nordic.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/thumb/v7e-m/nofp"  -LC:/Users/Gabriel/Desktop/nbps/zephyrproject/zephyr/samples/lorawan/class_a/build/zephyr  -lgcc  -Wl,--print-memory-usage  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no- [... cut ...] ' follow
Memory region         Used Size  Region Size  %age Used
           FLASH:       58712 B         1 MB      5.60%
            SRAM:       13152 B       256 KB      5.02%
        IDT_LIST:          88 B         2 KB      4.30%
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files (x86)\CMake\bin\cmake.EXE' --build 'C:\Users\Gabriel\Desktop\nbps\zephyrproject\zephyr\samples\lorawan\class_a\build'`

GABS9718, 2021-01-08:
Maybe the problem to link is related with those warnings, where is said that Lora was assigned with yes but got no. Why is this happening?

sophiekovalevsky, 2021-01-10:
@GABS9718 I built the application with 70ac3387ceb1c6ce05a40731a0a5c26a2d8c8539 using nrf52dk_nrf52832 board with the following symbols: 

```
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n

CONFIG_LOG=y
CONFIG_SPI=y
CONFIG_SPI_0=y
CONFIG_SPI_NRFX=y
CONFIG_GPIO=y
CONFIG_PRINTK=y

CONFIG_LORA=y
CONFIG_LORAWAN=y
CONFIG_LORAMAC_REGION_US915=y
CONFIG_LORA_SHELL=y
CONFIG_LORA_SX12XX=y
CONFIG_HAS_SEMTECH_RADIO_DRIVERS=y
CONFIG_HAS_SEMTECH_LORAMAC=y
```

One thing that you can try is remove the build directory and rebuild with pristine option enabled, I guess removing the build directory is kind of important before actually issuing the following: 

 `west build -p always  -b <your-board-model>`

Try that out, hope that works.

GABS9718, 2021-01-11:
I actually was removing the build folder to be sure that this would have influence. Do u have some another tip that I can try? Do u think that it can be related with the settings in the VS Code with the includePath directory? I'm really lost

sophiekovalevsky, 2021-01-11:
@GABS9718, I don't know about setting up Zephyr with VS Code, however, what I can think that you can do is just trying to compile the application using only west, if that doesn't work, I guess you need to find another option. But, I already tested with west, so, if all steps are okey, you should not have any issues.

GABS9718, 2021-01-17:
@Mani-Sadhasivam do u have some guess? I have been trying to fix it for a while, but I still just get the same result already writen in this topic.

Mani-Sadhasivam, 2021-01-17:
@GABS9718 The issue is mostly related to your Kconfig setting. Please use the below diff:

```
diff --git a/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts b/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
index 1417d75e96..bf744777a4 100644
--- a/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
+++ b/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
@@ -120,6 +120,7 @@
 		sw1 = &button1;
 		sw2 = &button2;
 		sw3 = &button3;
+		lora0 = &lora0;
 	};
 };
 
@@ -192,6 +193,21 @@ arduino_i2c: &i2c0 {
 	sck-pin = <31>;
 	mosi-pin = <30>;
 	miso-pin = <40>;
+
+	lora0: sx1276@0 {
+		status = "okay";
+		compatible = "semtech,sx1276";
+		reg = <0>;
+		label = "sx1276";
+		reset-gpios = <&gpio0 13 GPIO_ACTIVE_LOW>;
+		dio-gpios = <&gpio0 1 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
+			<&gpio0 2 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
+			<&gpio0 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
+			<&gpio0 4 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
+			<&gpio0 5 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
+		spi-max-frequency = <1000000>;
+		power-amplifier-output = "pa-boost";
+	};
 };
 
 &spi2 {
diff --git a/samples/lorawan/class_a/prj.conf b/samples/lorawan/class_a/prj.conf
index ec54d78f51..d9f33d944c 100644
--- a/samples/lorawan/class_a/prj.conf
+++ b/samples/lorawan/class_a/prj.conf
@@ -1,7 +1,16 @@
 CONFIG_LOG=y
 CONFIG_SPI=y
+CONFIG_SPI_1=y
+CONFIG_SPI_NRFX=y
 CONFIG_GPIO=y
 CONFIG_LORA=y
 CONFIG_LORA_SX12XX=y
 CONFIG_LORAWAN=y
-CONFIG_LORAMAC_REGION_IN865=y
+CONFIG_LORAMAC_REGION_US915=y
+CONFIG_SHELL=y
+CONFIG_LORA_SHELL=y
+
+CONFIG_USE_SEGGER_RTT=y
+CONFIG_RTT_CONSOLE=y
+CONFIG_UART_CONSOLE=n
+CONFIG_PRINTK=y
```

Also note that for getting lora to work, you need to choose the `power-amplifier-output` or `rfo-enable-gpios` or `pa-boost-enable-gpios` as per your board design.

GABS9718, 2021-01-17:
Hi @Mani-Sadhasivam, thanks for the help. I have tried what u said, but I still get an error. Do u think that is related with the above print? I got this when I did what u said.

[image]

Mani-Sadhasivam, 2021-01-18:
@GABS9718 Did you modify the sample application? There seems to include error which might due to wrong env settings or due to your change in samples/lorawan/class_1. But this error is not related to the change I shared. With the above diff, I could get the build to complete successfully.

GABS9718, 2021-01-19:
@Mani-Sadhasivam No, I did not. I just added in the device tree the node and the aliases for the lora. Everything else is the same as in the last zephyr version in master. I have add some print screens that show this.

[image]

[image]

[image]

[image]

[image]

Now, how my environment variables are setted in my PC. 

[image]

And the error

[image]

Do u have some other guess? I'm stuck in this for a while and idk what is going on. I have followed the instructions in the documents and everything else in this topic. I guess is something simple that I still could not find.

closed 2021-04-05 by github-actions[bot]