zephyrproject-rtos/zephyr #31168: Wrong linker option syntax for printf and scanf with float support

ArseniSoitu, 2021-01-07:
https://github.com/zephyrproject-rtos/zephyr/blob/712ff3298b2d9230a51ef60ba55467d719df3067/lib/libc/newlib/CMakeLists.txt#L37

Hello.

Seems that space need to be inserted between "-u" and "_printf_float"
Same on line 38

Thank You

pabigot, 2021-01-11:
Since that syntax has been in place for three years, it's unlikely to be wrong.  Are you getting compilation errors?

ArseniSoitu, 2021-01-12:
> Since that syntax has been in place for three years, it's unlikely to be wrong. Are you getting compilation errors?

Hmm. Tried to use sprintf() to convert float to string on stm32f413zh with code:
`char value[32];`
`voltage = 100.1f;`
`n = sprintf(value,"%4.1f", voltage);`
Result was `0.000`(don't remember actual number of zeros) in `value`.
But now after full rebuild I can't reproduce this bug.

I apologise for inconvenience

pabigot, 2021-01-12:
Thanks for the update.

closed 2021-01-12 by pabigot