zephyrproject-rtos/zephyr #31169: kconfig configuration (prj.conf) based on zephyr version

stoberblog, 2021-01-07:
I want to set a kconfig based on zephyr version but cannot find any details on how to do this. I think the kconfig system allows if statements, but I cannot find anywhere that would then allow me to do this based on the Zephyr version.

My reason for wanting this, is that it seems some of the power management configurations have changed names over the last month or so (eg CONFIG_SYS_PM_STATE_LOCK to CONFIG_PM_STATE_LOCK and CONFIG_SYS_POWER_DEEP_SLEEP_STATES to CONFIG_PM_DEEP_SLEEP_STATES). I want my code to work on older versions of Zephyr so my colleagues can update independently of when I update.

How would I achieve this?

Subsequently, how do I also use macros to determine the Zephyr version in code?, as similar to above, SYS_POWER_STATE_DEEP_SLEEP_1 has changed to POWER_STATE_DEEP_SLEEP_1

In terms of version I have installed of Zephyr, I updated Zephyr via git pull yesterday, and I had last updated early December. so between these two times is when the configuration names have changed.

martinjaeger, 2021-01-08:
Usually you would not work with the Zephyr master branch in production, but build your application firmware based on a specific release. This will also make sure that you and your colleagues will always have the same Zephyr version (branch specified in `west.yml` file).

If you need specific features that are only in master, I'd recommend to create a fork and cherry-pick those features to the previous release and point `west.yml` to that fork.

stoberblog, 2021-01-11:
I feel however this would still be useful feature to have such as for example code. The code I'm writing currently is example code that will be then used by others write the actual application on top of it. It will also likely be used by multiple projects which may be running differing versions of Zephyr and as such the code needs to cope with the different APIs

martinjaeger, 2021-01-11:
I'd probably disagree. If you are lucky, only some Kconfig names were changed (as you described above). But most probably also the code behind the API has changed, so this will lead to unpredictable behavior.

If the examples are upstreamed to main Zephyr repository you can just go back and forth in the git history to get a sample for an older version of Zephyr. And if you use west properly in your own out-of-tree samples you can do the same. Always run `west update` after `git checkout` and it will get you the Zephyr version you need.

closed 2021-03-28 by github-actions[bot]