zephyrproject-rtos/zephyr #31189: Add support for relaxed MMIO access routines

eugenecohen, 2021-01-08:
On architetures like aarch32 and aarch64 memory barriers i.e. `__DMB()` are present in the current implementation of MMIO access functions such as sys_read32 and sys_write32.

For many types of MMIO interactions these barriers have an adverse performance impact and are not necessary.  The performance impact has been observed to be most severe with looping PIO data transfer operations.

Other OSes like Linux support an exapnded MMIO access API providing ordered access i.e. writel and relaxed acces i.e. writel_relaxed.

To address this we can create additional MMIO access routines such as sys_read32_relaxed and sys_write32_relaxed and then over time update drivers to use these new routines.

eugenecohen, 2021-01-08:
@galak @nashif

nashif, 2021-01-19:
@eugenecohen you are welcome to submit a PR.

@carlocaione  @ioannisg FYI

carlocaione, 2021-01-19:
Thank you for pointing this out. @eugenecohen it's not clear to me if you have already code that you want to push otherwise I can take care of that.

eugenecohen, 2021-01-20:
Thanks - I wanted to see if you were in agreement with the idea before preparing patches.  I'll go ahead and prepare them now.