zephyrproject-rtos/zephyr #31191: Samples: TF-M: Enable NS test app(s)

microbuilder, 2021-01-08:
When performing a TF-M build, an optional NS test application can also be generated, based on: https://git.trustedfirmware.org/TF-M/tf-m-tests.git/tree/

This repo also includes PSA FF API tests (cf.: https://github.com/ARM-software/psa-arch-tests). PSA API FF certification of the RTOS requires sending the output of these tests results for the board(s) being used in the certification.

A sample application enabling easy building and flashing/running of these unit tests and of the PSA FF API tests will facilitate board certification, and may also be useful in CI to verify that all secure services and features are behaving as expected.

microbuilder, 2021-01-20:
Thinking about this, does a sample app make sense since there really isn't a code component here on the Zephyr side?

We could perhaps just document how to flash the NS binary that is generated as part of the reguar TF-M build (unless it's disabled), and explain that the PSA API FF tests are available there for certification or verification?

ioannisg, 2021-04-23:
Has this now been done @microbuilder , based on @oyvindronningstad 's last PR?

microbuilder, 2021-04-23:
The FF APIs aren't done, but aren't needed for PSA certification, which was the main goal of this issue, so I'll close this.

closed 2021-04-23 by microbuilder