zephyrproject-rtos/zephyr #31193: CMake integration

KKoovalsky, 2021-01-08:
Hello!

I wanted to start work with Zephyr as my main RTOS but unfortunately I abandon my efforts.

The main problem is that I have a current project whose build system is based on CMake and it works pretty well. I wanted to switch from another RTOS ecosystem to Zephyr since it's also built on top of CMake and its driver list is quite long.

The problem is that Zephyr doesn't work well with CMake. Let me list few problems which made me turn Zephyr down.

Order of the `find_package` and `project` calls is mandatory. Normally `find_package(Zephyr)` must be called before `project(...)` command. Zephyr sets up all the toolchain related paths and this is why this order is needed, among others, I guess. Why not rely on CMake's toolchain files functionality and let user choose the toolchain on his own? Zephyr could provide toolchain files on his own that set the initial C flags and others. I think that it could simplify the build system complexity. The compiler flags and all the toolchain related boilerplate could be defined as cmake targets. This also makes the project really application-centric. In this form the coupling to the Zephyr project is quite high unfortunately.

When defining multiple static library targets for which I would like to have proper compiler flags provided, I can't really do it from CMake, because Zephyr doesn't define any target that has compiler flags attached to it. Instead `COMPILER_OPT` is used, which is a string that is appended to the compiler invocation.

If you really want to be application-centric don't define executable targets (like the `app`) inside the Zephyr CMake module. It is preferred to define library or interface targets, so the user could link to it easily. 

I couldn't use `target_compile_features`. When I used `set_target_properties(... CXX_STANDARD ...` as a workaround the flags has been ignored by the setup. This means that I can't rely on any CMake feature because I am not really sure whether I have to enable the same function from the menuconfig.

My suggestion is to rework the build system step by step.

yashi, 2021-09-29:
(partially related to #8827 #36558)

`check_include_file()` doesn't work in module's CMakeLists.txt either. because module it doesn't know where to look for, and it wrongly find build machine's header file. it can be prevented by doing
```cmake
set(CMAKE_SYSROOT ${SYSROOT_DIR})
```
But still we need to set `CMAKE_REQUIRED_INCLUDES` and others to make `check_include_file()` work properly.

openthread module does something like this
```cmake
target_include_directories(ot-config INTERFACE
    $<TARGET_PROPERTY:zephyr_interface,INTERFACE_INCLUDE_DIRECTORIES>)
target_include_directories(ot-config SYSTEM INTERFACE
    $<TARGET_PROPERTY:zephyr_interface,INTERFACE_SYSTEM_INCLUDE_DIRECTORIES>)
target_compile_definitions(ot-config INTERFACE
    $<TARGET_PROPERTY:zephyr_interface,INTERFACE_COMPILE_DEFINITIONS>)
target_compile_options(ot-config INTERFACE
    $<TARGET_PROPERTY:zephyr_interface,INTERFACE_COMPILE_OPTIONS> -fno-builtin)
```
This allows us to build module code with Zephyr's compiler options.

Unfortunately, `check_include_file()` needs `CMAKE_REQUIRED_INCLUDES` to be set or flags given as additional param.

tejlmand, 2021-11-03:
> Why the order of calls:

Please find some explanation here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/application/index.rst#creating-an-application

> Why Zephyr CMake scripts perform complex actions to mutate CC, CXX variables to use the Zephyr's toolchain?

This has to do with the old migration from KBuild as @SebastianBoe has described but also the way Zephyr handles the compilers.
Zephyr will test the compiler for certain features and then cache the result with a unique compiler id in a user cache.
CMake can also test the compiler but the result of a test will only be stored in the build cache to avoid re-testing on subsequent CMake re-runs.
The Zephyr approach improves the speed significantly when building multiple samples with the same toolchain.

> Could CMake toolchain files be used instead? Could Zephyr rely on CMAKE_TOOLCHAIN_FILE?

There has been requests to support `CMAKE_TOOLCHAIN_FILE` also, and I do think it would make sense to have support for this in Zephyr. But I don't see Zephyr starting to provide CMake toolchain files this way.
One reason for that is the whole Kconfig / SoC / Boards infrastructure where the selection of a board determines the arch in use.
Requesting users to also point out a cross compiling toolchain in addition, with the risk of mismatching to rest of the system needs to be considered.

However, this should not prevent us from allow users with special wishes to specify CMAKE_TOOLCHAIN_FILE if they so like.
We just need to consider how such integration can happen safely and how it can be ensured in CI that such feature doesn't break later.

Also I have often seen developers putting compile flags in their toolchain files that are not essential for the toolchain / compiler itself, biut because it's convenient, such as release build flags, debug flags, etc.
Also some developers consider hardware flags as something to put into their toolchain file.
In Zephyr, all such build flags are controlled using Kconfig, so care must also be taken to ensure that it is clear what settings that will take precedence.

The short description above is part of the reason that other features have had higher priority until now.

SebastianBoe, 2021-11-03:
Zephyr used CMAKE_TOOLCHAIN_FILE at some point.

But the toolchain file must be specified each time on the command line, and to migrate to CMake we needed to be competitive with KBuild, which can be invoked as simply as "make".

So we dropped support for it.

I have no regrets about this decision.

tejlmand, 2021-11-03:
> But the toolchain file must be specified each time on the command line, and to migrate to CMake we needed to be competitive with KBuild, which can be invoked as simply as "make".

Since CMake 3.21 it can be specified as environment variable:
https://cmake.org/cmake/help/v3.21/envvar/CMAKE_TOOLCHAIN_FILE.html

Also with `west`, you may specify permanent CMake arguments, like:
`west config build.cmake-args -- -DCMAKE_TOOLCHAIN_FILE=<file>`
https://docs.zephyrproject.org/latest/guides/west/build-flash-debug.html#permanent-cmake-arguments 

and thus avoiding to type it on each new CMake invocation.

> I have no regrets about this decision.

me neither, but if someone finds it useful and valuable, as well as having the bandwidth to introduce it in a way that is compatible with existing Zephyr toolchain infrastructure, then I have no objections to such contribution.

tejlmand, 2021-11-03:
@KKoovalsky I just noticed your initial comment, so let me reply to some of your questions.

> When defining multiple static library targets for which I would like to have proper compiler flags provided, I can't really do it from CMake, because Zephyr doesn't define any target that has [...]

`COMPILER_OPT` is because this is a Kconfig configuration option that even allows you to append any compiler setting not already covered by Zephyr configuration system.

You should be able to find most commonly used flags that you require covered but as different compilers uses different flags for different compilers, there is a template for matching properties to flags, for example the template here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/compiler/compiler_flags_template.cmake
and then the gcc mappings:
https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/compiler/gcc/compiler_flags.cmake

This ensures that you can enable a single `CONFIG_SECURITY_CANARIES` in Kconfig and have it correctly mapped to the right set of choices for the toolchain in use:
https://github.com/zephyrproject-rtos/zephyr/blob/d5aa5c2a77f44ed7db227cbbf821275ad36c7524/CMakeLists.txt#L110-L111

mapping to:
https://github.com/zephyrproject-rtos/zephyr/blob/d5aa5c2a77f44ed7db227cbbf821275ad36c7524/cmake/compiler/gcc/compiler_flags.cmake#L155-L158
or
https://github.com/zephyrproject-rtos/zephyr/blob/d5aa5c2a77f44ed7db227cbbf821275ad36c7524/cmake/compiler/arcmwdt/compiler_flags.cmake#L169

depending on compiler in use and it's know capabilities (or tested capabilities if some versions of compiler might support a flag).

But of course not everything is covered.

If you want to do things purely in CMake and avoid Kconfig then you can take a look at the `zephyr_interface` interface library. I assume it's something like that you're looking for.

> I couldn't use target_compile_features.

This has been addressed here:
https://github.com/zephyrproject-rtos/zephyr/pull/39014

SebastianBoe, 2021-11-04:
I know you know this @tejlmand, but for others I would like to clarify that neither
west nor CMake 3.21 existed at that time.

tejlmand, 2021-11-04:
>  but for others I would like to clarify that neither west nor CMake 3.21 existed at that time.

absolutely correct.
I'm simply mentioning the new possibilities because if supporting CMake toolchain files would benefit the project overall, for example with less maintenance work, better toolchain support, easier integration, etc. then this single reason has been addressed externally.

But so far I still think the solution introduced with the KBuild re-write is superior as I describe in other comments.

boaks, 2022-05-25:
I'm struggling a little at an similar issue for an module. Applying [yashi/module-sample](https://github.com/yashi/module-sample) works well, but [AutoConf.cmake](https://github.com/eclipse/tinydtls/blob/develop/AutoConf.cmake) fails ins some case, e.g. adding "check_include_file(net/socket.h HAVE_NET_SOCKET_H)" doesn't work.

> Determining if the include file sys/socket.h exists failed with the following output:
> Change Dir: /home/achim/repos/zephyr/coaps-zephyr/coaps-zephyr/build_nrf9160dk_nrf9160_ns/CMakeFiles/CMakeTmp
> 
> Run Build Command(s):/usr/bin/ninja cmTC_40463 && [1/2] Building C object CMakeFiles/cmTC_40463.dir/CheckIncludeFile.c.obj
> FAILED: CMakeFiles/cmTC_40463.dir/CheckIncludeFile.c.obj 
> zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   -nostartfiles -nostdlib -isystem "zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include" -isystem "zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed" -Wl,--unresolved-symbols=ignore-in-object-files -Wl,--entry=0 -o CMakeFiles/cmTC_40463.dir/CheckIncludeFile.c.obj -c repos/zephyr/coaps-zephyr/coaps-zephyr/build_nrf9160dk_nrf9160_ns/CMakeFiles/CMakeTmp/CheckIncludeFile.c
> repos/zephyr/coaps-zephyr/coaps-zephyr/build_nrf9160dk_nrf9160_ns/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:10: fatal error: sys/socket.h: No such file or directory
>     1 | #include <sys/socket.h>
>       |          ^~~~~~~~~~~~~~
> compilation terminated.
> ninja: build stopped: subcommand failed.
> 

Adding:

`find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})`

results just in a different error:

> CMake Error at repos/zephyr/coaps-zephyr/zephyr/cmake/kconfig.cmake:295 (add_custom_target):
>   add_custom_target cannot create target "config-twister" because another
>   target with the same name already exists.  The existing target is a custom
>   target created in source directory
>   "repos/zephyr/coaps-zephyr/coaps-zephyr".  See documentation
>   for policy CMP0002 for more details.
> Call Stack (most recent call first):
>   repos/zephyr/coaps-zephyr/zephyr/cmake/app/boilerplate.cmake:544 (include)
>   repos/zephyr/coaps-zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
>   repos/zephyr/coaps-zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
>   repos/zephyr/coaps-zephyr/modules/lib/tinydtls/zephyr/CMakeLists.txt:24 (find_package)

In my case, it's only one header, so I fixed it. But it would be nice, if that works as well for modules.

KKoovalsky, 2024-02-14:
Hey!

I guess after such time the initial problem may be fixed with a different approach, or by adapting to the zephyr's build system requirements. :)

I haven't also used Zephyr for a while now, so I can't really give any input here. ;)

Thanks for the support anyway!

closed 2024-02-20 by marc-hb

canatella, 2024-12-31:
This would still be a nice to have regarding integration with Conan (see https://github.com/zephyrproject-rtos/zephyr/issues/31745) and probably other projects. I resorted to copy paste the compile flags used by Zephyr in a Conan profile where I could just have pointed to the toolchain file if it was there. Then I discovered that `find_zephyr` does it self end up calling `project` which created some recursive error with cmake if I included it in any subproject.

I understand where you're coming from, but when we recently started our new project, we thought "cool, Zephyr is CMake based, it'll be easy to integrate with our current code". Then I was disappointed to see it's yet another project working around CMake in its own unique, incompatibles ways. Reading the few issues around this here makes it clear there are actual reasons about this, and I understand that there's a lot of cmake code that would need to evolve, but it would be nice if there were some plan to move toward a more standard way of doing things. 

At least providing `CMAKE_TOOLCHAIN_FILE` support to enable compilation in other projects. No need for flashing or linking, but at the very least this would provide an easy way to build a static library to be integrated in the project later, without having to make it a Zephyr specific project so that if you're using conan, vcpkg or any other wrapper you can just drop `-DCMAKE_TOOLCHAIN_FILE=zephyr/my-board.cmake` and be good to go.

And then maybe by merging some features upstream? The global compiler flag cache is a nice thing and could maybe be useful to other projects.  For the compiler features,  maybe using `try_run` and `try_compile` with `cmake -C common-cache.txt` to populate the cache upfront could be a more standard way to do that? There's now preset files which could also be used to specify board specific variables. 

Just my 2 cents here. All in all, it's working. I'm just sad that in 2024, it's still complicated to integrate two unrelated C++ projects even though they both use CMake.

Best wishes for 2025 ;)

thorsten-klein, 2025-02-13:
I miss the ability to consume dependencies in Zephyr using `find_package()`, as is commonly done in CMake. Instead of relying on Zephyr modules for all third-party software, I want to encapsulate the build process of third-party dependencies separately from my main project build.  

**Note:** With `find_package()`, it is the user's responsibility to ensure that dependencies are correctly installed and discoverable by CMake (e.g., by setting the `CMAKE_PREFIX_PATH` environment variable). I believe Conan can be a great tool to manage this.  

I plan to experiment with creating Conan profiles and packaging all necessary Zephyr components (such as Zephyr and the Zephyr SDK) using Conan. This should allow me to easily build any third-party dependencies for Zephyr using `conan install`. 

To properly configure the environment before running `west build`, I can use the generated `conan_buildenv.sh`. If needed, `conan_toolchain.cmake` can also be used as the `CMAKE_TOOLCHAIN_FILE`.  

### What do you think?  
- Does this approach of including third-party dependencies via `find_package()` and Conan make sense?  
- Do you have any other suggestions for enabling `find_package()` in Zephyr builds?