zephyrproject-rtos/zephyr #31194: twister: using unsupported fixture causes an infinite loop during on-target test execution

PerMac, 2021-01-08:
**Describe the bug**
The bug can be observed when the following conditions combine:
- a test requiring a fixture 
- the test does not have harness: console or ztest 
- the test is called for hardware that does not have the fixture defined.

Even though a test requires a fixture in its yaml, twister will get in an infinite loop waiting for the device to become ready. I think the logic in `handle()` (line 689 in twisterlib.py) is wrong:
```
        while not self.device_is_available(self.instance):
            logger.debug("Waiting for device {} to become available".format(self.instance.platform.name))
            time.sleep(1)
```
The function `device_is_available()` checks if the fixture exists in the device and returns False. Then the 1 sec sleep is applied and the check for the fixture repeats. This happens until the timeout termination. 
If the test defines harness as console or ztest then the test is skipped correctly. If none of these harnesses is used then we end up in the loop. 

I am not sure if it is a real error. I found it when I was trying to figure out why #30134 was not getting filtered (now harness: ztest is added to the yamls in PR and fixture is skipped correctly).

PerMac, 2021-01-08:
@nashif related question: Why in `testcase_runnable()` (line 1761 of twisterlib.py) fixture requirement is checked only if testcase.harness in [ 'console', 'ztest']?

nashif, 2021-01-24:
> @nashif related question: Why in `testcase_runnable()` (line 1761 of twisterlib.py) fixture requirement is checked only if testcase.harness in [ 'console', 'ztest']?

because is the only way we have to deal with test output from fixtures

nashif, 2021-01-24:
a harness_config containing a fixture always needs a fixture in the yaml file. So this is more of a schema dependency validation issue and less of a logical issue I think.

nashif, 2021-01-30:
dont think this is a bug, it is just bad test configuration.

closed 2021-01-30 by nashif