zephyrproject-rtos/zephyr #31217: Multi-threading flash access is not supported by flash_write_protection_set()

nvlsianpu, 2021-01-11:
**The problem**
When two or more threads writes to flash device it is possible that flash protection will be enabled by one of threads despite another thread(s) still needs it disabled.

**Impact**

flash_write() or flash_erase() operation might fail unexpectedly while flash device is written or erased from multiple thread. This will be race condition.

**Solution**

Remember that below code is allowed:
``` C
flash_write_protection_set(flash_dev, false);
flash_erase(flash_dev, page_offset, page_size);

flash_write_protection_set(flash_dev, false);
flash_write(flash_dev, offset, data, sizeof(data));

flash_write_protection_set(flash_dev, true); // enable is recommended

```

Count flash protection disable/enable operation (with thread instance awareness, a thread might only increase/decrease counter by one).
Re-enable protection only once is zero.

nvlsianpu, 2021-01-11:
cc @anangl @carlescufi

de-nordic, 2021-02-24:
@nvlsianpu I think that stream_flash has similar problem on higher level.

armandciejak, 2021-03-08:
@nvlsianpu Your proposed solution would not work for flash chip that automatically enables write protection after write or erase command.

I have a custom board with such a flash chip and I'm impacted by this bug. The only solution I found so far is to hack the driver so that the lock is acquired by `flash_write_protection_set(flash_dev, false);` and released by `flash_erase` or `flash_write`. `flash_write_protection_set(flash_dev, true);` is a no-op for such flash.

armandciejak, 2021-03-08:
IMHO the only proper way to address this problem is to change the flash API. Removing the `flash_write_protection_set` function and changing `flash_erase` and `flash_write` so that they acquire the lock, disable the write protection do the erase/write operation, enable the write-protection and finally release the lock.

nvlsianpu, 2021-03-08:
@armandciejak 
> @nvlsianpu Your proposed solution would not work for flash chip that automatically enables write protection after write or erase command.

I think you understand me bad. I put the code sample in the description (same here https://docs.zephyrproject.org/latest/reference/peripherals/flash.html#c.flash_write_protection_set) - this code sample reflect usage with the automatic flash protection your mentioned, that why ther are dedicated flash disable calls before each write/erase operations. Solution I suggested will work for this case as well.

API change you mentioned is also possible solution worth to be considered.

Laczen, 2021-03-08:
@nvlsianpu, the thread awareness seems unnecessary complex. Maybe it is better to make `flash_write_protection_set` a `nop` and to include this in `flash_write` and `flash_erase`.

nvlsianpu, 2021-03-08:
@Laczen Agree, that will be much simpler but requires modification in each driver. I would go this way. Probably need to first present this on the API meeting.

carlescufi, 2021-03-09:
API meeting 9th March 2021

Conclusion is to go with the approach described as Alternative 1: deprecate `flash_write_protection_set ()` and unprotect/protect the flash on each call to `write` or `erase`.

PR #33216 (2021-03-29): drivers/flash: deprecation of flash_write_protection_set() API
The API is deprecated as it was decided to build in the flash protection
service into write and erase procedures.
This is solution chosen for fix following issue:
  When two or more threads writes to flash device it is
  possible that flash protection will be enabled by one of
  threads despite another thread(s) still needs it disabled.

fixes #31217 

- [x] spi_flash_at45 (already ready)
- [x] soc_flash_nrf (already ready)
- [x] nrf_qspi_nor
- [x] spi_nor
- [x] soc_flash_lpc
- [x] soc_flash_mcux
- [x] soc_flash_rv32m1
- [x] soc_fl [... cut]
closed 2021-03-29 by nashif