zephyrproject-rtos/zephyr #31219: newlib printk float formatting not working

gdf8gdn8, 2021-01-11:
**Describe the bug**
printk for foating point value not working.
printf format expected also in serveral drivers LPS25HB.

**Code**
```
float f= 123,456712356;
printk("%.6f\n",f);
```

**Output**
```
%.6f
```

**Expected behavior**
````
123,456712
````

**Environment (please complete the following information):**
 - OS: ArchLinux, Ubuntu 20.4
 - Toolchain 
 -- zephyr-sdk-0.11.4 
 -- west v0.8.0
 -- zephyr v2.3.0-2198

ArseniSoitu, 2021-01-11:
Hello. May be the reason is https://github.com/zephyrproject-rtos/zephyr/issues/31168 bug? Struggled with it several days ago.

pabigot, 2021-01-11:
`printk` uses a Zephyr-internal formatting utility now; you don't need to select newlib for that unless you're using standard C functions like `printf()`.  See [this documentation](https://docs.zephyrproject.org/latest/reference/misc/formatted_output.html).  You'll want `CONFIG_CBPRINTF_FP_SUPPORT=y`.

pabigot, 2021-01-11:
Ah; I see you seem to be using a Zephyr version before 2.4.0 was released.  `printk` didn't support FP back then.  The solution I recommended is available in the current development version and will be in 2.5 in about a month.

gdf8gdn8, 2021-01-12:
> Ah; I see you seem to be using a Zephyr version before 2.4.0 was released. ~`printk` didn't support FP back then.~ The solution I recommended is available in the current development version and will be in 2.5 in about a month.

It was fixed in v2.4.0. I updated to v2.4.0.

pabigot, 2021-01-12:
Great; thanks for the update.

closed 2021-01-12 by pabigot