zephyrproject-rtos/zephyr #31229: Full control of the connection process

BOZHENG001, 2021-01-11:
**Is your enhancement proposal related to a problem? Please describe.**
Hi, Zephyr community,

I am experiencing some difficulties when trying to have full control of the connection process. I am using 2 nRF52840DK boards to build a BLE connection. The zephyr version is 2.4.99. The samples I am using are `central_hr` and `peripheral_hr`.

What I am trying to do is to have full control of the connection process, such as connection interval, packet length, number of packets per connection interval, etc. I am able to change the connection interval, and the packet length from the peripheral (by changing the code in file "hrs.c"), but I don't know how to send data from central to peripheral. I try to use functions `bt_gatt_write`, but my nRF sniffer always shows "write not permitted". Besides, is it possible to control the number of packets per interval? For example, in each connection interval, there are always six packets transmitted (three from central, three from peripheral).

Hope you have some ideas.

**Describe the solution you'd like**
1. I would like to know how to use `bt_gatt_write` correctly on the central side.
2. I would like to know if I can control the packet number in each connection interval. If yes, how?

Thalley, 2022-01-24:
@BOZHENG001 Do you still have issues, or can this be closed?

BOZHENG001, 2022-01-24:
@Thalley This can be closed. Thanks.

closed 2022-01-24 by Thalley