zephyrproject-rtos/zephyr #31235: CortexM vector table relocation is incorrect when non XIP

GrixaYrev, 2021-01-11:
**Describe the bug**
In first, everything bellow is actual for case CONFIG_CPU_CORTEX_M_HAS_VTOR=y.
CortexM vector table relocate to SRAM start when CONFIG_XIP is not defined (see `arch\arm\core\aarch32\prep_c.c:50`).
But vector table can be placed with some offset, for example in i.MX RT.
So it is not possible to use non XIP application with i.MX RT boot header.

**To Reproduce**
1. i.MX RT1052
2. CONFIG_NXP_IMX_RT_BOOT_HEADER=y
3. CONFIG_XIP=n
4. Runtime error

**Expected behavior**
Definition `VECTOR_ADDRESS` is set to `_vector_start` in `arch\arm\core\aarch32\prep_c.c:48` if CONFIG_XIP=y.
But `_vector_start` is declared in `arch\arm\core\aarch32\vector_table.ld`, which in turn is added to linker script in `CMakeLists.txt` unconditional.
So it is possible to set `VECTOR_ADDRESS` equal `_vector_start` unconditionally also (if CONFIG_CPU_CORTEX_M_HAS_VTOR=y, of course).

**Impact**
Showstopper

**Environment:**
 - Commit SHA a7135bca17177122cfc70deb2dff4277c0108b72

ioannisg, 2021-02-01:
Will take a look

ioannisg, 2021-02-02:
Looks like the case with CPU_HAS_VTOR=y and XIP=n has not been changed, since it was originally introduced long time ago, in https://github.com/zephyrproject-rtos/zephyr/pull/5988/

ioannisg, 2021-02-02:
@GrixaYrev is there some specific in-tree sample or test you want to run with XIP=n

ioannisg, 2021-02-02:
@GrixaYrev  https://github.com/zephyrproject-rtos/zephyr/pull/31872 , please, see if this simple fix makes sense.

GrixaYrev, 2021-02-02:
@ioannisg is not, I use my custom project. In additional VTOR it is need to set custom MPU regions, because SRAM is not executable by default in `zephyr\soc\arm\common\cortex_m\arm_mpu_regions.c`.

ioannisg, 2021-02-02:
> @ioannisg is not, I use my custom project. In additional VTOR it is need to set custom MPU regions, because SRAM is not executable by default in zephyr\soc\arm\common\cortex_m\arm_mpu_regions.c.

OK, fine but this is another issue - not part of the ticket here.
(FYI vector table fetches do not suffer from MPU settings)

I wanted to check with you whether #31872 is in the right direction, in your opinion.

GrixaYrev, 2021-02-02:
> @GrixaYrev #31872 , please, see if this simple fix makes sense.

@ioannisg, I have tested `ioannisg:arch_arm_cortex_m_non_xip_vector_fix`, it's OK. VTOR have actual value now, i have checked with jLink debugger on i.MX RT1052.

PR #31872 (2021-02-03): arm: cortex_m: fix vector table relocation in non-XIP builds
When VTOR is implemented on the Cortex-M SoC, we can
basically use any address (properly aligned) for the
vector table starting address. We fix the setting of
VTOR in prep_c.c for non-XIP images, in this commit,
so we do not need to always have the vector table be
present at the start of RAM (CONFIG_SRAM_BASE_ADDRESS)
and allow for extra linker sections being placed before
the vector table section.

Fixes #31235 (@GrixaYrev pls, confirm)
closed 2021-02-03 by nashif