zephyrproject-rtos/zephyr #31253: lis3dh driver support is confusing

thedjnK, 2021-01-12:
**Describe the bug**
The ST lis3dh sensor device does appear to be supported but using the lis2dh driver, however the way it's been implemented is incredibly confusing and has wasted about an hour of my time this morning trying to figure it out, I'm not the only person that finds this confusing as noted in the original change https://github.com/zephyrproject-rtos/zephyr/pull/12373

The issue is that the lis2dh driver can be used for multiple devices, but there is a lis3dh driver binding which falls flat with the following build message:

    zephyr/drivers/sensor/lis2dh/lis2dh.c:353:2: warning: #warning "LIS2DH driver enabled without any devices" [-Wcpp]
      353 | #warning "LIS2DH driver enabled without any devices"
          |  ^~~~~~~

To get it to work, you must change to the lis2dh binding, my personal opinion is that this is annoying and bad, the lis3dh binding should be enough for the driver to work with it.
I haven't tested the other boards but I can see 96b_wistrio and rak5010_nrf52840 both only specify lis3dh bindings so I assume those actually do not work either.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure a board with an lis3dh using:

    lis3dh@18 {
    	compatible = "st,lis3dh";
    	label = "LIS3DH";
    	reg = <0x18>;
    };

2. Configure any project to use board and LIS2DH sensor
3. Build, it will output an error about no lis2dh sensors being present
4. Change DTS line:
    	compatible = "st,lis2dh";
5. Rebuild, error vanishes and device communication works

**Expected behavior**
The sensor to work out the box with the lis3dh binding

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Windows 7
 - Toolchain: GNU GCC from ARM (version 9.3.1, 2020 Q2 update)
 - Commit SHA or Version used: f4a26837c8cb6d39f3307193068c3163b20ac584

pabigot, 2021-01-19:
As long as there's a separate binding for `st,lis3dh` that's supposed to be supported by the lis2dh driver, I think that driver should iterate over lis3dh devices as well as lis2dh ones.  That would require repeating the generation loop with `#define DT_DRV_COMPAT st_lis3dh` in scope, assuming that you don't get two binding defines for the same device when the compatible is:

    compatible = "st,lis3dh", "st,lis2dh";

which is the way to solve this if lis3dh doesn't have any significant differences in binding content.  In that case just delete the `st,lis3dh` binding and convert the in-tree devicetree files to use the above compatible, which makes clear the real device but that it's compatible with the one we have a driver for.

avisconti, 2021-01-20:
> As long as there's a separate binding for `st,lis3dh` that's supposed to be supported by the lis2dh driver, I think that driver should iterate over lis3dh devices as well as lis2dh ones. That would [...]

Yes, this is the way it is expected to be done in DTS :

compatible = "st,lis3dh", "st,lis2dh";:

The LIS2DH and LIS3DH are register compatible. So same driver should work fine.

avisconti, 2021-03-22:
I'm planning to review this driver using STMEMSC i/f. Not sure exacvtly whether it makes sense to reuse same driver for all register compatible devices (lis2dh/lis3dh/lis2dh12/...) or create separate ones.

adamfowleruk, 2021-04-04:
I'm currently testing this with the LIS2DE12 sensor for a project we have in Herald. I'll let you know how I progress.

adamfowleruk, 2021-04-04:
I have confirmed the LIS2DH driver works with the LIS2DE12 sensor also. Sample app and custom board definition for our device (currently actually a nRF52833DK until our hardware is produced) can be found here: https://github.com/theheraldproject/zephyr-devices. Having ran the sample and manually checking the lis2dh.h settings with the LIS2DE12 datasheet, they are clearly compatible. My next task is adding double tap detection which I cannot currently see in the LIS2DH driver... Hopefully it's a relatively simple thing for me to add...

I've noticed that even copying the sample's config for floating point print support that it does not work. I've worked around that in my main.c.

Our code is here: https://github.com/theheraldproject/zephyr-devices

carlescufi, 2021-05-19:
@thedjnK as per the comments posted in this issue, do you think it can be closed?

closed 2021-08-19 by github-actions[bot]

zacck, 2025-08-13:
I wonder if this is really resolved?
On the `stm32f4_disco` that  has the `LIS3DSHTR` using the following overlay  with the `samples/sensor/lis2dh`

```

&spi1 {
	status = "okay";
	pinctrl-0 = <&spi1_sck_pa5 &spi1_miso_pa6 &spi1_mosi_pa7>;
	pinctrl-names = "default";
	cs-gpios = <&gpioe 3 GPIO_ACTIVE_LOW>;

	lis3dshtr: lis3dshtr@0 {
		compatible = "st,lis3dh", "st,lis2dh";
		reg = <0>;
		spi-max-frequency = <10000000>;
		status = "okay";

		irq-gpios = <&gpioe 0 GPIO_ACTIVE_HIGH>,  /* INT1 on PE0 */
		            <&gpioe 1 GPIO_ACTIVE_HIGH>; /* INT2 on PE1 */

		int1-gpio-config = <3>;  /* INT1 level high */
		int2-gpio-config = <3>;  /* INT2 level high */
	};
};

&gpioe {
	status = "okay";
};

&i2s3 {
	status = "disabled";
};

&i2s2 {
	status = "disabled";
};

/* Pin control for SPI1 */
&pinctrl {
    spi1_sck_pa5: spi1_sck_pa5 {
        pinmux = <STM32_PINMUX('A', 5, AF5)>;
    };

    spi1_miso_pa6: spi1_miso_pa6 {
        pinmux = <STM32_PINMUX('A', 6, AF5)>;
    };

    spi1_mosi_pa7: spi1_mosi_pa7 {
        pinmux = <STM32_PINMUX('A', 7, AF5)>;
    };
};

```

I get the following log
```
Device lis3dshtr@0 is not ready
[00:00:00.005,000] <err> lis2dh: Invalid chip ID: 3f

*** Booting Zephyr OS build v4.2.0-1490-ga5e687e46ac7 ***
```

Indeed `0x3f` is the correct CHIP ID for the onboard mems sensor.  Additionally the `LIS3DSHTR` needs some initial register configuration that I can't see happening on the current driver.

zacck, 2025-08-17:
Dug into this a little bit and I am beginning to doubt this driver works even with this lis2dh

first off when I allow the `0x3f` address in polling mode I just get constant data see below

```
#1 @ 6 ms: x -14.134176 , y 10.552752 , z -1.177848
[00:00:00.005,000] <inf> lis2dh: fs=2, odr=0x4 lp_en=0x0 scale=9576
*** Booting Zephyr OS build v4.2.0-1746-ge09fe269d78a ***
#2 @ 2011 ms: x -14.134176 , y 10.552752 , z -1.177848
#3 @ 4016 ms: x -14.134176 , y 10.552752 , z -1.177848
#4 @ 6021 ms: x -14.134176 , y 10.552752 , z -1.177848
#5 @ 8026 ms: x -14.134176 , y 10.552752 , z -1.177848
#6 @ 10031 ms: x -14.134176 , y 10.552752 , z -1.177848
#7 @ 12036 ms: x -14.134176 , y 10.552752 , z -1.177848
#8 @ 14041 ms: x -14.134176 , y 10.552752 , z -1.177848
#9 @ 16046 ms: x -14.134176 , y 10.552752 , z -1.177848
#10 @ 18051 ms: x -14.134176 , y 10.552752 , z -1.177848
```

The somewhat good news is that given time on a different run the data above may change. 

When I try to enable the triggers the driver break's down even further  with this overlay 
```
&spi1 {
    status = "okay";
    pinctrl-0 = <&spi1_sck_pa5 &spi1_miso_pa6 &spi1_mosi_pa7>;
    pinctrl-names = "default";
    cs-gpios = <&gpioe 3 GPIO_ACTIVE_LOW>;

    lis3dshtr: lis3dshtr@0 {
        compatible = "st,lis3dh", "st,lis2dh";
        reg = <0>;
        spi-max-frequency = <1000000>;
	status = "okay";

	irq-gpios = <&gpioe 0 GPIO_ACTIVE_HIGH>;

    };
};

&gpioe {
    status = "okay";
};
```

and this conf 

```
CONFIG_STDOUT_CONSOLE=y
CONFIG_SPI=y
CONFIG_SENSOR=y
CONFIG_LIS2DH=y
CONFIG_LIS2DH_TRIGGER_GLOBAL_THREAD=y
CONFIG_LOG=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y
CONFIG_CBPRINTF_FP_SUPPORT=y
``` 

I get this error 

```
Sampling at 1 Hz
Failed to set trigger: -134
[00:00:00.005,000] <inf> lis2dh: fs=2, odr=0x4 lp_en=0x0 scale=9576
*** Booting Zephyr OS build v4.2.0-1746-ge09fe269d78a ***
[00:00:00.006,000] <err> lis2dh: trigger_set DRDY int not supported
```

If I add a drydy gpio rather than irq-gpio

the build fails as such 

```
-- Found BOARD.dts: /home/zacck/zephyrproject/zephyr/boards/st/stm32f4_disco/stm32f4_disco.dts
-- Found devicetree overlay: /home/zacck/zephyrproject/zephyr/samples/sensor/lis2dh/app.overlay
devicetree error: 'drdy-gpios' appears in /soc/spi@40013000/lis3dshtr@0 in /home/zacck/zephyrproject/zephyr/build/zephyr/zephyr.dts.pre, but is not declared in 'properties:' in /home/zacck/zephyrproject/zephyr/dts/bindings/sensor/st,lis2dh-spi.yaml
CMake Error at /home/zacck/zephyrproject/zephyr/cmake/modules/dts.cmake:306 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

Call Stack (most recent call first):
  /home/zacck/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:131 (include)
  /home/zacck/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/zacck/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

````

It's as though the binding and the driver code have drifted. It would be nice to know if the issue is only with lis3dh  or even with lis2dh