zephyrproject-rtos/zephyr #31254: Extended advertising with one advertising set fails after the first time

StevenGhe, 2021-01-12:
**Describe the bug**
A clear and concise description of what the bug is.
I'm using an extended advertising set with one event (BT_LE_EXT_ADV_START_PARAM(0,1)). However if I call "bt_le_ext_adv_start" again with the same parameters, I get an error: (see picture below)

<err> bt_hci_core: Failed to start advertiser
[INIT] failed bt_le_ext_adv_start (err -5)  ( <-my own printk log)

Am I supposed to stop advertising or create a new extended advertising set each event?
The .sent callback in the "bt_le_ext_adv_cb" is called the first time, which makes me believe the advertising has stopped. 
Looking at the documentation, I see nothing else I can do?
I use an Adafruit Feather with a Nordic nRF52840 chip.

What have you tried to diagnose or workaround this issue?
Tried a lot of different methods of triggering this single advertising event. Nothing really changed
Multiple devices have the same behaviour

**To Reproduce**
Steps to reproduce the behavior:
I've supplied a main.c with some configuration files to test the problem in a .zip below.
The output generated on my device(s) is in the picture below.

**Expected behavior**
A clear and concise description of what you expected to happen.
Advertising should happen each time. Not just the first time.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
I'm quite sure I could use (BT_LE_EXT_ADV_START_PARAM(0,0) and stop the advertising myself given I stop it before the second event happened. So not too bad, but I think it should be a efficient using the provided function.

**Logs and console output**
[image]
[snippet.zip](https://github.com/zephyrproject-rtos/zephyr/files/5802087/snippet.zip)

**Environment (please complete the following information):**
 - OS: Win10
 - Toolchain: Zephyr
 - Commit SHA or Version used:
 1e61d3e

joerchan, 2021-01-12:
@cvinayak This seems to be an issue with the controller. The advertiser cannot be restarted after it has been terminated in the controller by a timeout or number of events limit.
The advertising set terminated event has been given to the host.

A simple test in the shell:
```
uart:~$ bt init
Bluetooth initialized
uart:~$ bt adv-create conn-nscan ext-adv
Created adv id: 0, adv: 0x20002800
uart:~$ bt adv-start
Advertiser[0] 0x20002800 set started
uart:~$ bt adv-stop
Advertiser set stopped
uart:~$ bt adv-start num-events 4
Advertiser[0] 0x20002800 set started
Advertiser[0] 0x20002800 sent 4
uart:~$ bt adv-start
Failed to start advertising set (-5)
[00:00:53.906,311] <wrn> bt_hci_core: opcode 0x2039 status 0x0c
[00:00:53.906,341] <err> bt_hci_core: Failed to start advertiser
```

cvinayak, 2021-01-12:
@joerchan Is the issue with non-connectable non-scannable extended advertising too?

joerchan, 2021-01-12:
@cvinayak No, that seems to work:
```
uart:~$ bt adv-create nconn-nscan ext-adv
Created adv id: 0, adv: 0x20002800
uart:~$ bt adv-start num-events 4
Advertiser[0] 0x20002800 set started
Advertiser[0] 0x20002800 sent 4
uart:~$ bt adv-start
Advertiser[0] 0x20002800 set started
```

cvinayak, 2021-01-12:
@joerchan ok. Thank you. Connectable was added recently, will add a connectable advertising test, investigate and send  a PR.

StevenGhe, 2021-01-12:
I'm sorry but I wasn't using connectable advertising? Only Identity and Extended options are set.

cvinayak, 2021-01-12:
@joerchan There is already a bsim test that covers starting connectable advertising after the terminate due to max events. (https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/bluetooth/bsim_bt/bsim_test_advx/src/main.c#L236) The test is passing in CI. I also added another start advertising with max events, advertising start without duration/max events and the test still passed for me.

updated test patch file: 
[0001-tests-Bluetooth-bsim-Added-advertising-start-after-t.zip](https://github.com/zephyrproject-rtos/zephyr/files/5803388/0001-tests-Bluetooth-bsim-Added-advertising-start-after-t.zip)

There must be something wrong running on target, will continue investigation...

cvinayak, 2021-01-13:
This is a bug in the controller, the implementation to release auxiliary channel PDU resources is missing when advertising terminates due to max events or duration. Relates to https://github.com/zephyrproject-rtos/zephyr/pull/26338

carlescufi, 2021-01-14:
Setting priority as low since this is a WIP feature in the controller.

carlescufi, 2021-01-21:
@StevenGhe this is caused by the current implementation of Extended Adv missing parts of it, it will be fixed as part of the ongoing Ext Adv work, but not in time for Zephyr 2.5.

StevenGhe, 2021-01-21:
Any ETA? Are we speaking of weeks/months?

My thesis is due +- May so not having such key BLE feature is pretty huge for me.

joerchan, 2021-01-21:
@StevenGhe I don't know the ETA, but in the meantime you could use the SoftDevice controller in nRF Connect SDK, which has support for this feature.

legoabram, 2021-02-16:
I have a similar problem which I believe is related. An advertising set will also fail to restart if a connection was made through that set. In that instance, BT_HCI_ERR_MEM_CAPACITY_EXCEEDED is returned instead. The error starts in `ull_adv.c` at line 978 when calling `ll_rx_alloc`. It looks like extended advertising sets don't always cleanup after themselves.
[prj.zip](https://github.com/zephyrproject-rtos/zephyr/files/5991628/prj.zip)

PR #32754 (2021-03-10): Bluetooth: controller: Fix auxiliary PDU disable
When Extended Advertising terminated due to duration or
maximum number of events, the auxiliary PDU scheduling
is now correctly stopped.

Fixes #31254.
commit 0fbf31a1 (2021-03-10): Bluetooth: controller: Fix auxiliary PDU disable

When Extended Advertising terminated due to duration or
maximum number of events, the auxiliary PDU scheduling
is now correctly stopped.

Fixes #31254.
closed 2021-03-10 by nashif