zephyrproject-rtos/zephyr #31258: whatch dog (WWDT) timeout calculation for STM32 handles biggest timeout and rollover wrong

sandroklarer-pingpong, 2021-01-12:
**Describe the bug**
The calculation for the timeout in function wwdg_stm32_convert_timeout(...) handles rollover and max timeout wrong, and as soon as i set the timeout over the limit, I do not get an error as expected. 

**To Reproduce**
General: 
stm32f412 system clock 100MHz
CONFIG_CLOCK_STM32_APB1_PRESCALER=2 -> 50MHz

Reproduce on a stm32f412g_disco eval board:
add in stm32f412g_disco.dts
```
&wwdg {
status = "okay";
};
```
add in prj.conf
```
CONFIG_WWDG_STM32=y

west build -c -p auto -b stm32f412g_disco samples/drivers/watchdog
west flash
```
**Expected behavior**
Expect an accurate timeout setting, by invoking wdt_install_timeout(..) and an error, if I'm more then 10 % off my setting.

**Impact**
Until this bug is fixed, we have to set the timeout max value to 41, instead of 42.

**Logs and console output**
Added two line to log the calculated timeout and desired timeout:
```
diff --git a/drivers/watchdog/wdt_wwdg_stm32.c b/drivers/watchdog/wdt_wwdg_stm32.c
index 82b899a5c1..d882588df4 100644
--- a/drivers/watchdog/wdt_wwdg_stm32.c
+++ b/drivers/watchdog/wdt_wwdg_stm32.c
@@ -189,6 +189,8 @@ static int wwdg_stm32_install_timeout(const struct device *dev,
     wwdg_stm32_convert_timeout(dev, timeout, &prescaler, &counter);
 
     calculated_timeout = wwdg_stm32_get_timeout(dev, prescaler, counter);
+    LOG_WRN("Desired WDT: %d us",timeout);
+    LOG_WRN("Set WDT: %d us",calculated_timeout);
     if (!(IS_WWDG_PRESCALER(prescaler) && IS_WWDG_COUNTER(counter) &&
           IS_WWDG_TIMEOUT(timeout, calculated_timeout))) {
         /* One of the parameters provided is invalid */
```
diffent runs:
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 100000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 409 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 84000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 163 us

[00:00:00.014,000] <wrn> wdt_wwdg_stm32: Desired WDT: 83000 us
[00:00:00.014,000] <wrn> wdt_wwdg_stm32: Set WDT: 41943 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 64000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 22282 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 43000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 1310 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 42000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 655 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 41000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 41287 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 40000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 40632 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 20000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 20316 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 10000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 10485 us

[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Desired WDT: 1000 us
[00:00:00.013,000] <wrn> wdt_wwdg_stm32: Set WDT: 1310 us

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK
 - Zephyr zephyr-v2.4.0-1-g32e4d7006338 

**Additional context**
I think it should start the calculation with the smallest counter/divider to get the most accurate timeout.
The test constant WWDG_TIMEOUT_ERROR_MARGIN should not be 100ms, a percentage would be nicer:
#define WWDG_TIMEOUT_ERROR_MARGIN   __TIMEOUT__)   (__TIMEOUT__ / 10)

I did a fast fix for us, maybe this helps: 
```
diff --git a/drivers/watchdog/wdt_wwdg_stm32.c b/drivers/watchdog/wdt_wwd
index 82b899a5c1..12f04733a2 100644
--- a/drivers/watchdog/wdt_wwdg_stm32.c
+++ b/drivers/watchdog/wdt_wwdg_stm32.c
@@ -56,10 +56,10 @@ LOG_MODULE_REGISTER(wdt_wwdg_stm32);
  */
 
 #define ABS_DIFF_UINT(a, b)  ((a) > (b) ? (a) - (b) : (b) - (a))
-#define WWDG_TIMEOUT_ERROR_MARGIN   (100 * USEC_PER_MSEC)
+#define WWDG_TIMEOUT_ERROR_MARGIN(__TIMEOUT__)   (__TIMEOUT__ / 10)
 #define IS_WWDG_TIMEOUT(__TIMEOUT_GOLDEN__, __TIMEOUT__)  \
     (ABS_DIFF_UINT(__TIMEOUT_GOLDEN__, __TIMEOUT__) < \
-     WWDG_TIMEOUT_ERROR_MARGIN)
+    WWDG_TIMEOUT_ERROR_MARGIN(__TIMEOUT_GOLDEN__))
 
 static void wwdg_stm32_irq_config(const struct device *dev);
 
@@ -113,7 +113,7 @@ static void wwdg_stm32_convert_timeout(const struct d
 {
     uint32_t clock_freq = wwdg_stm32_get_pclk(dev);
     uint8_t divider = 0U;
-    uint8_t shift = 3U;
+    uint8_t shift = 0U;
 
     /* Convert timeout to seconds. */
     float timeout_s = (float)timeout / USEC_PER_SEC;
@@ -122,18 +122,18 @@ static void wwdg_stm32_convert_timeout(const struct
     *prescaler = 0;
     *counter = 0;
 
-    for (divider = 8; divider >= 1; divider >>= 1) {
-        wwdg_freq = ((float)clock_freq) / WWDG_INTERNAL_DIVIDER / divide
+    for (divider = 0; divider <= 3; divider ++) {
+        wwdg_freq = ((float)clock_freq) / WWDG_INTERNAL_DIVIDER / (1 << 
         /* +1 to ceil the result, which may lose from truncation */
         *counter = (uint32_t)(timeout_s * wwdg_freq + 1) - 1;
-        *counter |= WWDG_RESET_LIMIT;
+        *counter += WWDG_RESET_LIMIT;
         *prescaler = shift << 7;
 
         if (*counter <= WWDG_COUNTER_MAX) {
             break;
         }
-
-        shift--;
+        shift++;
+        *counter = WWDG_COUNTER_MAX; //set to max, if no smaller counter
     }
 }
```

erwango, 2021-01-19:
@sandroklarer-pingpong Txs for the issue and proposed fix.
It seems we miss a bit in the proposed fix:
```
+    for (divider = 0; divider <= 3; divider ++) {
+        wwdg_freq = ((float)clock_freq) / WWDG_INTERNAL_DIVIDER / (1 << 
```
Can you complete the line (to save some time ) ?

sandroklarer-pingpong, 2021-01-21:
+    for (divider = 0; divider <= 3; divider ++) {
+        wwdg_freq = ((float)clock_freq) / WWDG_INTERNAL_DIVIDER / (1 << divider);

erwango, 2021-01-21:
I'm fine with the proposed change in `wwdg_stm32_convert_timeout`.
About the `WWDG_TIMEOUT_ERROR_MARGIN` macro, it makes sense indeed to compare result in a relative fashion. Though I wonder if we should keep `ABS_DIFF_UINT`.
I'm not a user of this api, but I wonder if we shouldn't rather guarantee that set timeout is  higher than requested timeout.
`watchdog.h` seems to state the opposite (but it depends how you understand it, I rather think it is not clear enough):
```
 * @param max Upper limit of watchdog feed timeout in milliseconds.
```
Though, I guess that from application point of view, you need to ensure that watchdog won't timeout before the requested timeout (that I guess the application is able to monitor). Setting a timeout shorter to this application requested timeout would be harder to monitor (even with a known tolerance).
I am seeing things correctly ?

sandroklarer-pingpong, 2021-01-22:
` * @param max Upper limit of watchdog feed timeout in milliseconds.`
There are two limits for this watch dog. This just says, that this parameter is the upper/max one. As a user, I want to get the best match for my set limit, so I do not care if it's a bit more or less in time. (My opinion.)

PR #31700 (2021-02-02): stm32: window watchdog: Fix timeout computation
Fix Timeout computation on stm32 window watchdog.
Additionally perform few changes in order to enable and ease tests in CI.

Fixes #31258
closed 2021-02-02 by nashif